wled/WLED #4105: Brightness slider inaccurate

blake-nouribekian, 2024-08-17:
### What happened?

The brightness slider does not work correctly. I get to full brightness on the LED's when I get to 19/255 brightness. Going any higher does not increase the brightness.

### To Reproduce Bug

Not sure if this causes any issue but I have "Enable automatic brightness limiter" on and set to 1000mA. Turn on the LED's and slide the brightness in the upper right hand corner.

### Expected Behavior

The brightness mapping for the slider will change the LED's brightness from 1 to 255

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0-b3 (build 2405180)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-08-17:
Disable brightness limiter. Brightness works as expected.

blake-nouribekian, 2024-08-17:
Will disabling the limiter cause LED's to burn out if the power supply is too much then? Seems like with the limiter on that the slider should be adjusted.

blazoncek, 2024-08-18:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
Closing.

closed 2024-08-18 by blazoncek

softhack007, 2024-08-18:
> By disabling the limiter, that would let the LED strip draw more power than the PSU can handle if it is lower than the LED strip. Wouldn't that cause an issue?

The brightness limiter is  meant to protect your wires. If you use thin wires and the current is too high, then wires will get extremely hot. It also protects your esp board, in case you power the led strip directly from the 5V pins on the board.

* https://kno.wled.ge/basics/top5_mistakes/

* https://wled-calculator.github.io/
*