wled/WLED #4109: Atom Echo install not working

modem7, 2024-08-19:
### What happened?

When installing on an [atom echo](https://shop.m5stack.com/products/atom-echo-smart-speaker-dev-kit), I get an unhandled exception.

### To Reproduce Bug

Install via https://install.wled.me

### Expected Behavior

For it to work

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.4

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40146887  PS      : 0x00060b30  A0      : 0x80146854  A1      : 0x3ffb1d70  
A2      : 0x00000010  A3      : 0x00000002  A4      : 0x00060f20  A5      : 0x00000001  
A6      : 0x00060f20  A7      : 0x00000000  A8      : 0x80146854  A9      : 0x3ffb1d50  
A10     : 0x00000000  A11     : 0x00060f23  A12     : 0x00060f2Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40146887  PS      : 0x00060b30  A0      : 0x80146854  A1      : 0x3ffb1d70  
A2      : 0x00000010  A3      : 0x00000002  A4      : 0x00060f20  A5      : 0x00000001  
A6      : 0x00060f20  A7      : 0x00000000  A8      : 0x80146854  A9      : 0x3ffb1d50  
A10     : 0x00000000  A11     : 0x00060f23  A12     : 0x00060f20  A13     : 0x3ffb1efc  
A14     : 0x000000ff  A15     : 0x3ffb0060  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x0000004c  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x40146884:0x3ffb1d70 0x40146851:0x3ffb1d90 0x40144e56:0x3ffb1db0 0x400ee20b:0x3ffb1de0 0x400ee7f6:0x3ffb1e30 0x400f04f6:0x3ffb1e70 0x400f0632:0x3ffb1e90 0x400f06d1:0x3ffb1ec0 0x400eb4bd:0x3ffb1ee0 0x4010f889:0x3ffb1f20 0x40112226:0x3ffb1f40 0x401132ae:0x3ffb1f90 0x40132516:0x3ffb1fb0 0x4008b89e:0x3ffb1fd0

Rebooting...
```

### Anything else?

I have also tried the latest beta, but there was no log output when I did so (nor did it work)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-08-19:
Please open an issue with [WLED installer](https://github.com/Aircoookie/WLED-WebInstaller), not WLED.

softhack007, 2024-08-19:
Additionally, you could try this semi-official installer site, which sometimes works better for uncommon boards:

https://wled-install.github.io/

modem7, 2024-08-19:
Thank you very much!

I shall give those a whirl!

It's not a deal breaker as I have other boards, but might as well give it a try to use these up!