wled/WLED #4113: Can add more than 10 ddp lights to esp32 but fails to work correctly after

Gamerayers, 2024-08-21:
### What happened?

I am trying to control 11 ddp Esphome esp32 led bulbs which I have in a path. When you add more than 10 to LED preferences, everything looks great. The problem is that when you go back in, you cannot edit them and it appears only LEDs 1-10 control work and 11 does not. They are not a unique string, just individual lights.
[image]

### To Reproduce Bug

Configure 11 ddp bulbs using RGBW on an ESP32

### Expected Behavior

Allow for 11 individual bulbs to be turned into an led string.

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0-b4/2406290 

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-08-21:
Try recent sources. B4 has a known bug in LED settings.

Gamerayers, 2024-08-21:
> Try recent sources. B4 has a known bug in LED settings.

I just installed 0.14.4. It does not allow for adding any more than 10 ddp lights. Maybe that is the limit? Ideally I would love to add in all 11. Is the best solution to add a second esp running wled for the 11th light and sync it? With it only being 11 single LEDs, I question if the esp32 would be able to accomplish this.

blazoncek, 2024-08-21:
I think 10 *is* the absolute limit.

closed 2024-12-28 by github-actions[bot]