wled/WLED #4116: many segment's one relay

niqu1982, 2024-08-25:
### What happened?

Hi i read ton's of discussion and not find any solution.
It's possible to change little bit of configuiration relay?
When i dissable all segment's master switch is still on and big PSU still working - wasting electricity.
Is possible to made if all segment's off then big PSU is also off and when any segment on then big psu is on
I use dig quad with esp32

### To Reproduce Bug

turn on master switch but all segment's is off

### Expected Behavior

when i turn off segments then relay ( big PSU) go off

### Install Method

Binary from WLED.me

### What version of WLED?

WLED.15.0.-b4 also WLED 14

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-08-26:
There is no way currently to have such behaviour.
Closest it can get is Mult Relay usermod.

niqu1982, 2024-08-26:
So on this moment better to use 3 PSU and 3 different "controller"

p1otr, 2025-02-27:
That feature would be super useful for me as well. I guess I have to use ESPHome for now

niqu1982, 2025-02-28:
For this moment if You have home assistant: create group of light. Create automation to sync group to main power using blueprint Synchronize the on/off state of 2 entities.