wled/WLED #4120: Crashes when leds length is 3

Andro-Marian, 2024-08-27:
### What happened?

WLed crashes when setting some effects like Fireworks 1D

### To Reproduce Bug

1. Set the Length of leds to 3
2. Set the effect to Fireworks 1D

### Expected Behavior

To work fine.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.4, 0.15.0-b4

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
abort() was called at PC 0x4008f49c on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008a7b8:0x3ffb7710 0x4008ab15:0x3ffb7730 0x4008f49c:0x3ffb7750 0x4008f9b4:0x3ffb7770 0x4008600d:0x3ffb7790 0x4008603e:0x3ffb77b0 0x4008645d:0x3ffb77d0 0x4000beaf:0x3ffb77f0 0x40084081:0x3ffb7810 0x4018a2b1:0x3ffb7830 0x401a1ce9:0x3ffb7850 0x4019cf91:0x3ffb7870 0x4014dc73:0x3ffb7890 0x4014dcdb:0x3ffb78b0 0x4008c55e:0x3ffb78d0

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Ada
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-08-27:
Can you, please, use exception decoder and provide decoded output?
You will need to compile sources yourself, preferably using debug options.

Andro-Marian, 2024-08-27:
If you can help me with a guide on VSCode how to debug, because I already try and it was unsuccessful.

blazoncek, 2024-08-28:
Please use WLED [forum](http://wled.discourse.group) or Discord for that. It has been answered many times before.

Andro-Marian, 2024-08-28:
Backtrace: 0x4008a2c4:0x3ffb1cf0 0x4008a621:0x3ffb1d10 0x4008ede8:0x3ffb1d30 0x4008f300:0x3ffb1d50 0x40085add:0x3ffb1d70 0x40085b0e:0x3ffb1d90 0x40085f2d:0x3ffb1db0 0x4000beaf:0x3ffb1dd0 0x401b095f:0x3ffb1df0 0x401b0931:0x3ffb1e10 0x401265df:0x3ffb1e30 0x4010e9b5:0x3ffb1e70 0x40116844:0x3ffb1f50 0x40117236:0x3ffb1f90 0x40136fad:0x3ffb1fb0 0x4008c06a:0x3ffb1fd0
  #0  0x4008a2c4:0x3ffb1cf0 in invoke_abort at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:648
  #1  0x4008a621:0x3ffb1d10 in abort at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:648
  #2  0x4008ede8:0x3ffb1d30 in multi_heap_assert at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:380
      (inlined by) multi_heap_malloc_impl at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:432
  #3  0x4008f300:0x3ffb1d50 in multi_heap_malloc at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:321
  #4  0x40085add:0x3ffb1d70 in heap_caps_malloc at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:232
  #5  0x40085b0e:0x3ffb1d90 in heap_caps_malloc_default at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:232
  #6  0x40085f2d:0x3ffb1db0 in _malloc_r at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:37
  #7  0x4000beaf:0x3ffb1dd0 in ?? ??:0
  #8  0x401b095f:0x3ffb1df0 in operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc:50
  #9  0x401b0931:0x3ffb1e10 in operator new[](unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_opv.cc:32
  #10 0x401265df:0x3ffb1e30 in WiFiUDP::parsePacket() at C:\Users\DesktopTV\.platformio\packages\framework-arduinoespressif32\cores\esp32/Udp.h:46
  #11 0x4010e9b5:0x3ffb1e70 in handleNotifications() at wled00/udp.cpp:237
  #12 0x40116844:0x3ffb1f50 in WLED::loop() at wled00/wled.cpp:62
  #13 0x40117236:0x3ffb1f90 in loop() at C:/Shared/WLED-0.14.4/wled00/wled00.ino:20
  #14 0x40136fad:0x3ffb1fb0 in loopTask(void*) at C:\Users\DesktopTV\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:23
  #15 0x4008c06a:0x3ffb1fd0 in vPortTaskWrapper at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Andro-Marian, 2024-08-28:
[New Text Document (2).txt](https://github.com/user-attachments/files/16778392/New.Text.Document.2.txt)

Andro-Marian, 2024-08-28:
[New Text Document (2).txt](https://github.com/user-attachments/files/16778405/New.Text.Document.2.txt)

blazoncek, 2024-08-28:
From the crash dump you provided it can be seen that the crash happens in system libraries (WiFi UDP) during handleNotification().

handleNotification() is used if you have sync receive enabled and slave device listens for sync messages. I would suspect a rogue packet...

And BTW, please use 0.15 when submitting bugs. 0.14 will not receive updates any more and several bugs were fixed in 0.15 already.

Andro-Marian, 2024-08-28:
The Sync button is not activated. About that you say?

blazoncek, 2024-08-28:
> The Sync button is not activated. About that you say?

That button is for master aka sender. This is for slave aka receiver. It is always active but you can disable receiving by unchecking all receive groups.

Andro-Marian, 2024-08-29:
The 0.14 still crash when unchecking all the boxes from  Sync Interface.
Also, can't compile the 0.15
`> node tools/cdata.js 

node:internal/modules/cjs/loader:1227
  throw err;
  ^

Error: Cannot find module 'inliner'
Require stack:
- C:\Users\DesktopTV\Desktop\WLED-0_15\tools\cdata.js
    at Module._resolveFilename (node:internal/modules/cjs/loader:1224:15)
    at Module._load (node:internal/modules/cjs/loader:1050:27)
    at Module.require (node:internal/modules/cjs/loader:1310:19)
    at require (node:internal/modules/helpers:179:18)
    at Object.<anonymous> (C:\Users\DesktopTV\Desktop\WLED-0_15\tools\cdata.js:20:17)
    at Module._compile (node:internal/modules/cjs/loader:1480:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1564:10)
    at Module.load (node:internal/modules/cjs/loader:1287:32)
    at Module._load (node:internal/modules/cjs/loader:1103:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:168:12) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [ 'C:\\Users\\DesktopTV\\Desktop\\WLED-0_15\\tools\\cdata.js' ]
}

Node.js v22.1.0
*** Error 1`

Andro-Marian, 2024-08-29:
[0.15.txt](https://github.com/user-attachments/files/16798412/0.15.txt)
Also, why the 0.15 takes so long to rebot than 0.14

Andro-Marian, 2024-08-29:
I don't think is a network issue. Another use managed it to reproduce on discord.
And is only when setting the leds length below or equal to 3.
And when switching to Fireworks 3D or when setting to other effect when leaving the Fireworks 3D.
For my doesn't looks a network issue. If what that, then should crash no matter how many leds length is.

willmmiles, 2024-11-16:
Reproduced on my test setup, investigating now

willmmiles, 2024-11-27:
Fix is merged in 0_15.

PR #4289 (2024-11-18): Fix array overflow in exploding_fireworks
Attempt to allocate enough room for the "minimum" sparks; and ensure that we never overrun the allocated array size.

Fixes #4120
commit ce764803 (2024-11-25): Fix array overflow in exploding_fireworks

Attempt to allocate enough room for the "minimum" sparks; and ensure
that we never overrun the allocated array size.

Fixes #4120
closed 2024-11-27 by willmmiles