wled/WLED #4121: WLED 0.15 seven_segment_display_reloaded crashes ESP32-C3

KrX3D, 2024-08-30:
### What happened?

Hello,

i have an ESP32-C3, i just downloaded WLED 0.15 and enabled in the my_config file the usermod:

seven_segment_display_reloaded
with
#define USERMOD_SSDR

after flasing i cant connect to the gui and it seems the esp just hangs. i tried like 3 or 4 other ESP32-C3 and always the same.
when i dont enable  the usermod it works.

i got it to "work" by disabling those lines in the usermod_seven_segment_reloaded.h

line 244 and 247:
//umSSDRMask[i] = true;

line 262:
//umSSDRMask[i] = false;

and also line 255:
 //strip.setPixelColor(i, 0x000000);

all 4 need to be commented out or i cant connect to the gui

### To Reproduce Bug

- get WLED 0.15 and enabled in the my_config file the usermod: (also enable my_config  in the wled.h file)
- enable in the config the usermod seven_segment_display_reloaded with #define USERMOD_SSDR

### Expected Behavior

it should work like in 13.3 and 14.4

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.15.0.-b4 BUILD 2407070

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

KrX3D, 2024-08-31:
some more info:

the esp32-c3 where before working iwth wled 0.14.4 and there i had 3 led outpouts set with ws281x than i updated to 0.15 and i realized that the 3rd led output was missing. and i cant set it again like i mentioned here:

https://github.com/Aircoookie/WLED/issues/4122

now i reflashed 0.15-b4 via https://install.wled.me/ and all settings where reset.
i reflashed my self compiiled version and when i dont set up any led output i just need to comment out line 244 :  //umSSDRMask[i] = true;

the other 3 can be as intended. even when i set up led output 1 and 2

so maybe the 3rd output that went "missing" after updating to 0.15 caused this?

but this usermod is still broken since line 244 crashes it

rickdoq, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

blazoncek, 2024-08-31:
You will need to contact usermod creator or maintainer as WLED team does not support usermods unless personally involved.

KrX3D, 2024-08-31:
Hi,

im not really a github user. How do i know who created the usermod and how can i contact him?

also it is possible like i mentioned that after upgrading from 0.14 to 0.15 and one led output got missing (becausxe of not enough RTM channels like you mentioned on the other thread) that the 3rd led output was still "stored" and cause some problems, which got soloved after all userdata was deleted ? or do you think that was something else?

blazoncek, 2024-08-31:
Use Git blame feature (available on Github) to find out original author or current maintainer.
Git commit history is another way but too much cumbersome.

As for upgrading, backup settings and presets. If anything does not play as you expect, erase all settings and see if reconfiguring helps. If not, there might be a bug (0.15 is still a beta) or the feature has changed.

KrX3D, 2024-08-31:
ok i think i got it working. and it seems not to be a problem with the usermod, but it could be handled better by the usermode.

like i said i had 3 led outputs and when i switched to 0.15 only 2 led outputs are there and the third was missing, like you said bec ause of not enough RTM channels.
but the usermod was still set to use the 10 leds that the third led output had. 

and so the usermod was using those 10 leds from led output 3 which is missing. so for me it seems to be an array issue.

```
if (range) {
for(int i = lastSeenLedNr; i <= lednr; i++) {
	umSSDRMask[i] = true;
}
} else {
umSSDRMask[lednr] = true;
}
```

i removed the 10 leds which i set in the usermod config in the gui and now it seems to work

KrX3D, 2024-08-31:
i got it woking now and added some checks in the _setLeds function

im mostly checking for umSSDRLength . here is the improved code which works now even when i have set some leds that are not present:

`  void _setLeds(int lednr, int lastSeenLedNr, bool range, int countSegments, int number, bool colon) {
    if (number >= 0 && number < 11 && countSegments >= 0 && countSegments < 7) {
        if ((colon && umSSDRColonblink) || umSSDRNumbers[number][countSegments]) {
            if (range) {
                // Check if the LED index is within the valid range before accessing the array
                if (lednr < umSSDRLength && lastSeenLedNr < umSSDRLength) {
                    for (int i = lastSeenLedNr; i <= lednr; i++) {
                        umSSDRMask[i] = true;
                    }
                }
            } else {
                // Check if the single LED index is within the valid range before accessing the array
                if (lednr < umSSDRLength) {
                    umSSDRMask[lednr] = true;
                }
            }
        }
    }
  }`

softhack007, 2024-09-06:
@KrX3D I've fixed some potential array bounds violations and pointer errors in the usermod. Please check if it works for you.

Important: I'm not the maintainer of this usermod, and I have no idea what the usermod does. I've just applied a bandaid where it seemed reasonable.

KrX3D, 2024-09-08:
hi,

i tested your fix here:
https://github.com/Aircoookie/WLED/commit/340a9f8b762195b10f64ff42af9228e5140bb6b3

for now it seems to work, so thanks for that.

i will test it for the next 2-3 days and report back

closed 2024-09-10 by softhack007

KrX3D, 2024-09-11:
Hi, 

so its not completly solved. But its maybe not only the fault of this usermod.

What i did:

- Download a fresh copy of 0.15-b5
- enable in wled.h - #define WLED_USE_MY_CONFIG
- in my_config.h i added my ssid and my pass and also added - #define USERMOD_SSDR
- in platformio.ini i changed it to this: default_envs = lolin_s2_mini 

than i flashed a completly new esp32 s2 (i tried even others) and when i enable the usermod everything works but when i uncheck at 

/settings/leds  ---> Turn LEDs on after power up/reset:  and i restart the esp via gui or unplug the device it doesnt connect again. 
in device manager (windows) it is also not shown as COM Port

when i dont enable the usermod and i uncheck it, and restart the device it works.
also when i dont uncheck it and set everything up, it also works.

its just this combination that "breaks" it