wled/WLED #4122: WLED 0.15.0-b4 - 3rd LED output -> greyed out options

KrX3D, 2024-08-30:
### What happened?

on a fresh WLED 0.15.0-b4 when i add 2 led outpouts as WS281x as here:

[image]

and i want to add a 3rd one there a a lot of options greyed out and i cant select a 3rd WS281x:

[image]

### To Reproduce Bug

- Flash a fresh compiled version of WLED 0.15.0-b4 and add 2 LED Outputs as WS281x
- Add a 3rd led out and there are a lot of greyed out options:

[image]

### Expected Behavior

it should be possible to add a 3rd (and more) LED Output (WS281x etc) as it was working in WLED 0.14.4

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.15.0.-b4 BUILD 2407070

### Which microcontroller/board are you seeing the problem on?

ESP32-S2, ESP32-C3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-08-31:
The disabled types are intentional as they are disabled due to missing HW support (i.e. not enough RMT channels).

closed 2024-08-31 by blazoncek

KrX3D, 2024-08-31:
Hi, so this means that with an S2 C3 i cant add more ws281x than 2 led outputs? if yes what other esp32 could i use?

also it would be nice to see in the Gui WHY they are greyed out