wled/WLED #4125: Add Support for Relaying External Data Line Input

fordcrews, 2024-09-01:
Summary: I would like to request the addition of a feature in WLED that allows it to accept an input signal from the data line of external lights, such as Govee lights, on one of the ESP32's data pins. When this external signal is detected, WLED should relay it to the connected LEDs. If no signal is detected, WLED should revert to normal operation, controlling the LEDs as it typically would.

Use Case: This feature would be particularly useful for users who have existing smart lights (like Govee) and want to integrate WLED's advanced capabilities without losing the ability to use the original light controller. This would allow the external controller to take precedence when active, while WLED controls the LEDs when the external controller is off.

Proposed Implementation:

    Configure Input Pin:
        Allow users to configure one of the ESP32's GPIO pins as an input for receiving the data signal from the external light controller.

    Signal Detection:
        Implement logic to detect the presence of a signal on the configured input pin.
        If a signal is detected, bypass WLED's internal LED control and relay the signal directly to the LED strip.

    Signal Relaying:
        Relay the captured signal from the input pin to the output pin connected to the LED strip when the external controller is active.
        When no signal is detected, allow WLED to resume normal control over the LEDs.

    Fallback to Normal Operation:
        If the external controller is not active (i.e., no signal on the input pin), WLED should control the LEDs as usual, using its internal settings and effects.

Additional Considerations:

    Timing & Data Format: Care must be taken to ensure that the signal timing and data format from the external controller are compatible with WLED's control protocols.
    User Configuration: Users should have the ability to enable or disable this feature and configure the input pin via the WLED UI or configuration files.

Benefit: This feature would enhance WLED's versatility by allowing it to seamlessly integrate with existing smart lighting setups, providing users with more flexibility and control over their lighting environments.

Potential Challenges:

    Ensuring compatibility with different types of external controllers.
    Handling the timing and data format differences between various light controllers.

Conclusion: This feature would be a valuable addition to WLED, offering users the ability to bridge their existing smart light setups with WLED’s robust control capabilities. It would provide a smooth transition between different control sources and expand WLED's functionality.

softhack007, 2024-09-02:
Hi, 
I'm not sure how we could implement this proposal.

Electrical:
* ws2812 (and many other LEDs protocols) work at 5V signal level. ESP chips require 3.3V level, and normal boards are _not_ 5V tolerant. So attaching a 5V signal to an esp32 input pin might destroy the esp32, unless you add a level shifter or other means to reduce the signal to 3.3V.

Software:
* I don't know of any means how to "relay" (or forward) a signal from one pin to another pin. 
* If done in pure software, it would require sampling at more than 4x800khz, and a continuous and uninterrupted output at the same rate. This is almost impossible to achieve, considering that even a 1 _microsecond_ glitch would violate the ws2812 protocol and cause visible flickering.

Please also keep in mind that WLED uses the NeoPixelBus library for led driving. The WLED core team has no experience with inplementing high speed LED rotocols with hardware acceleration support.

As an alternative solution,  you could try to use pure hardware. A relay to bypass the esp32 output signal and use an external signal source might work. The relay could be triggered by a simple circuit that detects when a plug was inserted (the external controller ). I guess you would also need a diode to protect the esp32 output pin against reverse currents.

blazoncek, 2024-09-02:
> When this external signal is detected, WLED should relay it to the connected LEDs

Please describe in detail what "relay it to LEDs" means.

fordcrews, 2024-09-02:
I was figuring if it was ones that used 5V control a couple of diodes would drop it low enough to be safe.   I thought I had seen someone using gpio_matrix/mux to do something similar.   If this isn't doable, I guess I'll see if an NPN transistor and a 74HC157  can be used to detect data/switch control without need for this change.    

Sorry to have wasted your time.

k7bbr, 2024-09-02:
Your use case is quite common -- you want to control a string of lights
with a particular controller some of the time while being able to control
them with a WLED controller the rest of the time. Usually this is with
permanent lights like Govee, Trimlights, Gemstone, etc. Quite difficult to
do in software, but super easy to do with hardware, and use of the already
implemented relay function in WLED. The HC157 or LS157 line of multiplexers
(or any 2 line to 1 line data selector/multiplexer chip) will do the job.
Just tie the output of the relay line from WLED to the select pin of your
multiplexer. When it goes high, the WLED controller is selected. When it's
low, the data stream defaults to your other controller.

You can also get a pre-made multiplexer for LED data here:
https://wasatchpixels.com/product/wasatch-multiplexer/

-Brandon

softhack007, 2024-09-06:
Closing as "not planned". Its better to solve this with hardware (relay), because a software solution would be much more complicated.

closed 2024-09-06 by softhack007