wled/WLED #4130: AP always defaults back to WLED_000000

scarolan, 2024-09-06:
### What happened?

Attempted to rename the access point to something different, no matter what I try the access point fails back to WLED_000000.

I'm using an Athom.tech with 0.13.0-b6 preinstalled.  

### To Reproduce Bug

Connect to the access point.  Enter the WiFi setup page.  Put in a new name for AP SSID.  Save settings and reboot device.  

### Expected Behavior

Expected the new SSID I assigned to stay put.  Oddly the password does save.  It's just the SSID that won't.

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.0-b6

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-09-06:
Hi, actually athom have added their own code for naming the AP. The behaviour you describe is not happening with normal WLED. Please report the problem to athom - unfortunately the WLED core team cannot help you here.

In fact you could try to attach a serial-to-USB board and then flash your controller with a normal WLED firmware. However a few athom specific additions (IR remote, RF433) will not be available any more.

> Attempted to rename the access point to something different, no matter what I try the access point fails back to WLED_000000.
> 
> I'm using an Athom.tech with 0.13.0-b6 preinstalled.

scarolan, 2024-09-06:
Ah, thank you for clearing this up!  I was going crazy here wondering what I did wrong.

closed 2024-09-06 by scarolan