wled/WLED #4132: Correct mA/LED settings for WS2805 (RGB+WW+CW)?

Moonbase59, 2024-09-07:
### What happened?

WS2805 strips typically come as

- per 1 LED addressable at 5V
- per 3 LEDs addressable at 12V
- per 6 LEDs addressable at 24V

1, 3 or 6 LEDs are connected in series to one of the WS2805 outputs, respectively, and—according to the [datasheet](https://github.com/Aircoookie/WLED/files/13753418/Worldsemi-WS2805.EN.pdf)—the chip drives a typical 12 mA, or max. 14 mA output current on each of the R, G, B, W1 and W2 outputs.

**So what is the correct current (mA/LED) to set in WLED for the power calculation/current limiting to function properly?
5 x 12 mA = 60 mA, or 5 x 14 mA = 70 mA?**

Could the correct value be added to the selection dropdown for novices?

### To Reproduce Bug

Set (and connect) a WS2805 5-channel LED strip in v0.15.0-b4, try to figure out correct power settings.

### Expected Behavior

Have a suggestion for novices in the selection dropdown, and possibly document if WLED calculates using "typical" or "maximum" current consumption.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0-b4 (build 2406290)

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-09-07:
ABL (automatic brightness limiter) does not operate on CCT or RGBWWCW LED strips properly as the code internally does not work with 2 whites.
Also W channel is not part of the calculation of used current independently.

So even if one would want to have *correct* calculation it is currently impossible without complete code rewrite. And even then not all LED strips are made equal and actual used current may differ from calculation.

Adding additional dropdown options adds little value as unavailable current can be added via custom option.

Moonbase59, 2024-09-11:
Hm. Thanks, that is valuable information.

Do you think…

- this will come in a future version, sometime?
- I could get by not overdriving my poor little USB-C power supply by "artificially" specifying a higher custom mA/LED, for the time being?

blazoncek, 2024-09-11:
You can disable ABL entirely and use adequate PSU.
Please use on-line calculators to determine maximum current draw.

- most likely no, unless someone finds interest in doing so
- just stay safe, whatever you attempt

kalwinskidawid, 2024-10-05:
Hey,
@Moonbase59 How did you finally set it up? I have the same problem

Sawtaytoes, 2025-05-31:
I figured the default 55mA was too low. I'm also wondering which calculation makes sense, 12mA or 14mA. I'd err on the higher one.