wled/WLED #4135: Support more than 1 Wiz remotes

DarkSlice1, 2024-09-09:
**Is your feature request related to a problem? Please describe.**
I'm installing a coving with led as the main light source in a room. The users of this are not tech savvy and have been using the wiz remote to control the lighting.
They are looking for a wall mounted switch also - the Wiz smart button.
Wled does not support more than 1 Wiz control per wled device 

**Describe the solution you'd like**
Support of 2 wiz controllers per wled device 

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

Thank you for your ideas for making WLED better!

Erik-JanWind, 2025-01-09:
I am looking for this also

DedeHai, 2025-01-10:
@blazoncek how much would it take to add multiple? would it be as easy as adding another ID to the config and checking both upon ESP receive?

blazoncek, 2025-01-10:
How many? 2, 3 or 4? Or more? Where to stop?

The best way would be to implement pairing in a proper sense: start pairing in WLED and add as many as user needs. Best implemented with vectors. How much effort? Not trivial but possible.

Erik-JanWind, 2025-01-10:
For my particular need it's 8 remotes, but the timeline is very limited. (Need by end of February). But could be a great feature of it's expandable to 2-3 remotes.

blazoncek, 2025-01-10:
> but the timeline is very limited

Then perhaps invest time and implement for the rest of us. 😉

Erik-JanWind, 2025-01-10:
> > but the timeline is very limited
> 
> Then perhaps invest time and implement for the rest of us. 😉

If I could, I would, this is way beyond me skillset. I am only a Enduser with needs, asking small questions that probably entails mountains of work ... ;-)

closed 2025-05-19 by DedeHai