wled/WLED #4136: Reset Segments does not respect 'Make a segment for each output'

DedeHai, 2024-09-10:
### What happened?

when 'Make a segment for each output' is checked I would expect reset segments to set it to that default instead of just deleting all segments.

### To Reproduce Bug

check the setting, setup some segments and reset them.

### Expected Behavior

see 'What happened?'

### Install Method

Binary from WLED.me

### What version of WLED?

0.15 (but maybe also in 0.14, did not check)

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-11-22:
Might be good to unterstand what's going on before we release 0.15.0-RC1 https://github.com/Aircoookie/WLED/pull/4287

blazoncek, 2024-11-25:
This behaviour is present since introduction of *multistrip* in 2021.

netmindz, 2024-11-27:
If this isn't a regression, then I would vote to remove this as a blocking issue for the 0.15.0 release

softhack007, 2024-11-27:
> If this isn't a regression, then I would vote to remove this as a blocking issue for the 0.15.0 release

I agree

blazoncek, 2024-11-27:
https://github.com/user-attachments/assets/87658be2-c0f5-4466-8486-3f77b55f7d16

If you decide you need it.