wled/WLED #4143: Cannot compile anymore

NickDolfin, 2024-09-14:
### What happened?

yesterday, I could compile.
Now I get the following error:

PackageException: Package version 2.3.0+sha.00be156 doesn't satisfy requirements 2.2.1 based on PackageMetadata <type=library name=ESPAsyncWebServerWLED version=2.3.0+sha.00be156 spec={'owner': None, 'id': None, 'name': 'ESPAsyncWebServer', 'requirements': '2.2.1', 'uri': 'git+https://github.com/Aircoookie/ESPAsyncWebServer.git'}

### To Reproduce Bug

Install vscode, platformio and the latest wled source zip file.
just click build

### Expected Behavior

no errors perhaps?

### Install Method

Binary from WLED.me

### What version of WLED?

WLED-0.15.0.b4 and also WLED-0.15.0.b5

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Relevant log/trace output

```shell
Resolving esp32s3dev_8MB_opi dependencies...
Library Manager: Installing git+https://github.com/pbolduc/AsyncTCP.git @ 1.2.0
git version 2.46.0.windows.1
Cloning into 'C:\Users\nick\.platformio\.cache\tmp\pkg-installing-eosqjot5'...

Library Manager: AsyncTCP@1.2.0+sha.38b8475 has been installed!
Library Manager: Installing fastled/FastLED @ 3.6.0
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: FastLED@3.6.0 has been installed!
Library Manager: Installing makuna/NeoPixelBus @ 2.8.0
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: NeoPixelBus@2.8.0 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing git+https://github.com/Aircoookie/ESPAsyncWebServer.git @ 2.2.1
git version 2.46.0.windows.1
Cloning into 'C:\Users\nick\.platformio\.cache\tmp\pkg-installing-vgz50be_'...

PackageException: Package version 2.3.0+sha.00be156 doesn't satisfy requirements 2.2.1 based on PackageMetadata <type=library name=ESPAsyncWebServerWLED version=2.3.0+sha.00be156 spec={'owner': None, 'id': None, 'name': 'ESPAsyncWebServer', 'requirements': '2.2.1', 'uri': 'git+https://github.com/Aircoookie/ESPAsyncWebServer.git'}
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

willmmiles, 2024-09-14:
The version pin on the ESPAsyncWebServer library was bad -- it's trying to use the 'master' branch but expecting it to have a specific version number, which has now changed.   #4142 has the fix.

blazoncek, 2024-09-14:
Fixed

closed 2024-09-14 by blazoncek

NickDolfin, 2024-09-14:
Thanks guys for the updates.

After I did a full clean and build again.
The same error appears.
What Should I do?

willmmiles, 2024-09-14:
Use latest 0_15, or cherry pick 2264cc5d10aae1fff21bf34782a09ff5291c9c3d

NickDolfin, 2024-09-14:
Resolving esp32s3dev_8MB_opi dependencies...
Library Manager: Installing git+https://github.com/Aircoookie/ESPAsyncWebServer.git @ 2.2.1
git version 2.46.0.windows.1
Cloning into 'C:\Users\nick\.platformio\.cache\tmp\pkg-installing-8df237f6'...

PackageException: Package version 2.3.0+sha.00be156 doesn't satisfy requirements 2.2.1 based on PackageMetadata <type=library name=ESPAsyncWebServerWLED version=2.3.0+sha.00be156 spec={'owner': None, 'id': None, 'name': 'ESPAsyncWebServer', 'requirements': '2.2.1', 'uri': 'git+https://github.com/Aircoookie/ESPAsyncWebServer.git'}

NickDolfin, 2024-09-15:
Thanks for showing the cherry pick. I changed it by hand since I have no clue on how to use git.