wled/WLED #4144: Static IP on Ethernet Boards

Fonzi03, 2024-09-15:
This was posted here as a bug: https://github.com/Aircoookie/WLED/issues/3927
I see it was marked closed but I am uncertain why? It maybe should have been a feature request?

Just like Wifi, Ethernet should have the capability of putting in a static IP. 

As a note, same as mentioned in the bug above, a gateway should never be required but instead optional. 

If static IP capabilities are added for Ethernet, they should be separate from the Wifi settings as users may want them both configured.

netmindz, 2024-09-23:
I don't think WLED is really designed to handle being attached to multiple networks simultaneously, so single fixed IP should be ok

If you set the gateway IP as well as your device's IP do you see the static IP set correctly to the wired interface?

Fonzi03, 2024-09-24:
I cant test this any time soon as the board is in my attic and I havent gotten my network pathway up there yet. May be able to test in about 3 weeks or so.