wled/WLED #4149: 5ghz ESP32+ RTL8720 usermode

Negr92, 2024-09-18:
Я некоторое время изучал проект WLED. Я использую светодиод для управления светодиодами с помощью ARTNET и 1.31. иногда я замечаю помехи на 2,4 ГГц (когда я использую его на сцене). Помехи, вызванные другим оборудованием. Я хочу использовать WLED на частоте 5 ГГц. Существует очень мало модулей с поддержкой 5 ГГц. Мой выбор пал на Realtek RTL8720. Я планирую подключить RTL8720+ESP32 через UART. RTL8720 подключается к WIFI 5ghz и передает данные через UART на ESP32. Как вы думаете, можно ли это сделать в качестве ПОЛЬЗОВАТЕЛЬСКОГО РЕЖИМА?  P.S. Я не профессиональный программист. Я поклонник написания простых программ. WLED слишком сложный для меня. Я прошу помощи и совета.

willmmiles, 2024-09-19:
I don't think it's currently possible to proxy network access over the serial port; as far as I'm aware that's never been done with Espressif's TCP/IP stack.

You could potentially write a program for the RTL8270 to act as its own ARTNET/e1.31 endpoint and convert the messages to the Adalight protocol that WLED supports on the serial port.  It would be entirely independent of WLED, though.  If you found the Adalight protocol to be unsuitable, you could potentially implement your own as a usermod.

netmindz, 2024-09-23:
Depending on what control you are looking for, you might also be able to use wired DMX as the input from the other board - if you are using the in-built effects rather than realtime (dumb) mode