wled/WLED #4150: [Bug] Brightness is not persistent after saving the settings

Andro-Marian, 2024-09-19:
### What happened?

When going to LED Preferences and pressing Save, the brightness of LEDs are increased.

### To Reproduce Bug

1. Set a lower brightness of leds, like 10
2. Go to setting -> Led Preferences -> Click Save

The leds brightness are higher than normal.

### Expected Behavior

When pressing save to have the brightness set as the slider option.

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0-b4, b5

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Seams a REGRESSION from 0.14.4

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2024-09-25:
can confirm. this bug has been around for a while, so not something introduced recently.

softhack007, 2024-10-05:
I think @blazoncek fixed this in https://github.com/Aircoookie/WLED/commit/a4c49aa35e93d158033844f6d433fae56def8514 ?

blazoncek, 2024-10-05:
Yes, I did.
Closing as a duplicate of #4005

closed 2024-10-05 by blazoncek