wled/WLED #4151: [Feature] Allow using Presets when DDP,UDP data is received

Andro-Marian, 2024-09-19:
**Is your feature request related to a problem? Please describe.**
When the DDP data is enabled I cannot use Presets.
Presets that are used by some buttons to toggle the MultiRelay.

Or some presets that has a way to change the brightness of Leds faster.

Presets should be allowed on any case. Just preventing to change the Effects, Color.

**Describe the solution you'd like**
Allow using presets when DDP, UDP data is controlling the leds effect.

**Describe alternatives you've considered**
No exists.

**Additional context**
-

Andro-Marian, 2024-09-19:
What I that?

blazoncek, 2024-09-19:
Don't click!

netmindz, 2024-09-22:
That is a bit of an unusual use case. Presets are primarily designed for effects - when in "realtime" mode WLED is just in dump pass-thru mode of the supplied data to the pixels.

Can you make the changes you wish to control the Multi-Relay usermod though the API?

Andro-Marian, 2024-09-22:
Not sure how would you control the buttons with the API to toggle a relay.

blazoncek, 2024-09-23:
Just to clarify: When WLED is receiving external data (or live data, i.e. from xLights or another WLED device as a DDP virtual LED) it becomes "dumb" device. This is intended and will most likely not change.

There are options (in Sync Settings) that will allow you to still control WLED while it is receiving live data. Please explore them.
Of course there is always an option to "override" live data and get control of the UI by clicking on override.

Andro-Marian, 2024-09-23:
At least allow buttons to trigger presets.

Andro-Marian, 2024-10-08:
Who need this fix edit the wled.cpp and move the:
```  handlePresets();
  yield();
```

before:

```
if (!realtimeMode || realtimeOverride || (realtimeMode && useMainSegmentOnly))
```