wled/WLED #4153: Segment "update" hint not disappearing

Andro-Marian, 2024-09-22:
### What happened?

When clicking on the checkmark from a segment the hint will never disappear.
[image]
[image]

### To Reproduce Bug

Go to a segment and click on checkmark.

### Expected Behavior

To disappear. Wnd not not appear if the click is not holding.

### Install Method

Binary from WLED.me

### What version of WLED?

0.15

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Was working good on 0.14

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2024-09-22:
need more details. works fine in chrome browser.

Andro-Marian, 2024-09-22:
Not in firefox or the WLED App

softhack007, 2024-09-22:
> Screenshot_2024-09-22-09-40-50-779_ca cgagnier wlednativeandroid

Looks like the screenshot is from the [wled native](https://github.com/Moustachauve/WLED-Native-Android) app on android by @Moustachauve

softhack007, 2024-09-22:
Maybe we should remove this "update" hint ? I mean the behaviour of the checkmark seems obvious without the hint, right?

blazoncek, 2024-09-22:
Every HTML tag that has "title" attribute uses help pop-up. I think the original author may be @ajotanc but also @w00000dy is quite experienced with UI behaviour, if they can help.

PR #4253 (2025-02-15): Fix for #4153
Fixes #4153

The issue was caused by the `mouseover` event sometimes being triggered on button release for touch inputs. This resulted in the tooltip appearing but never closing.

I resolved the issue by only adding the `mouseover` and `mouseout` events on devices without touchscreens, 
while the `touchstart` and `touchend` events are loaded for devices with touchscreens.
closed 2025-02-15 by DedeHai