wled/WLED #4155: Usermod installation + web based flash tool

Gittegatt, 2024-09-22:
Hello,
usermods are very popular to unleash the full ability spectrum of WLED. ♥
In order to 'easify' the installing process of WLED with a specific usermod, may it possible to implement the feature of combining the installation of WLED + usermods avoiding to do a text-based choice of the usermods in the listfile and the self-compilation?

Users could do a usermods choice e.g. with a checkbox during the installation via the web based flash tool https://install.wled.me  ?
That would make it more user-friendly.

This solution request has been already posted by @henrygab on a [post at wled.discourse.group](https://wled.discourse.group/t/wled-build-with-all-usermods-integrated-selectable/2814/6?) 

Furthermore I faced the issue to be unable to update this self-compiled version with a usermod via the OTA built-in updater. Message: _Update failed! Please check your file and retry!_
But I guess this corresponds by using the self-compiled one. If yes, may it possible to make the OTA update possible?

Hats off and Regards.

DedeHai, 2024-09-23:
That is a great idea, we are currently just lacking someone who will write the code and test it across all systems and versions.

blazoncek, 2024-09-23:
Unfortunately usermods are written in C++ and need to be compiled during WLED compilation.
This means that if you want certain combination of usermods, all of those need to be compiled at the same time, producing a binary with them included.

There are over 60 usermods so the available combinations are virtually endless. Hence it is not possible to do what you request to have as many binaries as there are combinations.

We have considered the option to include most requested usermods in "standard" installation but no verdict has been achieved yet.

Gittegatt, 2024-09-23:
Okay,
so if this is such a unmanageable thing, I can imagine and understand this is too big.

Regards.

netmindz, 2024-09-23:
You might want to check out this tool that lets you do custom builds where you can select which usermods to include 

https://wled-compile.github.io/

Kalki2898add, 2025-07-13:
i downloaded the latest build of wled code and compiled it on platformio working fine , but how can i add usermods? i tried old method (worked on 0.14 version ) but not working new version. what to do?