wled/WLED #4156: Ability to disable mDNS

gaaat98, 2024-09-22:
I have 5 WLED devices on my network, and since I deployed the fifth one, I started having random reachability problems in my wireless LAN and various issues with the mDNS capabilities on other non-WLED devices. After months of investigation, I figured out that the problem is probably related to the mDNS traffic that WLED generates that is somehow """saturating""" the forwarding capabilities of my ISP-provided router, causing a failure of other mDNS services as well as other connectivity issues.

I do realize that this is not completely related to WLED but rather a very specific corner case, but I still wanted to know if it is possible to have the option to disable mDNS and what features rely on it. Thanks in advance!

netmindz, 2024-09-22:
mDNS is for the local network only, so there should be no forwarding at all of this traffic from your network to any other network.

Please supply packet capture using Wireshark if you believe that WLED is "flooding" your local network with mDNS

What is more likely is one or more of your WLED devices are using sleep mode, which then triggers routers to change the way that Multicast traffic is handled by an Access Point. 

Disable sleep mode and/or move your WLED devices to their own SSID/VLAN

If you still want to disable mDNS, then just remove the name - The config states "mDNS address (leave empty for no mDNS):"

closed 2024-09-22 by netmindz

gaaat98, 2024-09-22:
> What is more likely is one or more of your WLED devices are using sleep mode, which then triggers routers to change the way that Multicast traffic is handled by an Access Point.
> 
> Disable sleep mode and/or move your WLED devices to their own SSID/VLAN
> 
I will try this, thank you very much!

> If you still want to disable mDNS, then just remove the name - The config states "mDNS address (leave empty for no mDNS):"
>
Sorry, I feel very dumb now, I should have looked more at the settings 😅

eku, 2025-03-26:
@netmindz wrote:
> If you still want to disable mDNS, then just remove the name - The config states "mDNS address (leave empty for no mDNS):"

Although I set the value to empty as described from you, I still see Multicast mDNS packets being sent from WLED.

```
IN=wlan0 OUT= MAC=XX SRC=192.168.33.132 DST=224.0.0.251 LEN=191 TOS=0x00 PREC=0x00 TTL=255 ID=5657 PROTO=UDP SPT=5353 DPT=5353 LEN=171
```