wled/WLED #4161: WLED-AP doesn't appear on ESP32-S3-Tiny after flash

me-dex, 2024-09-26:
### What happened?

Attempted to flash latest 0.15.0-b5 beta onto an [ESP32-S3-Tiny](https://www.waveshare.com/wiki/ESP32-S3-Tiny) via the built in DFU serial connection. Used both esptool.py and the web flasher. Flashing succeeded with both tools but AP failed to appear after reset. Flashed both with/without bootloader and no success in either case. Latest Tasmota working on the board so the chip/RF path does work

### To Reproduce Bug

Perform flash of 0.15.0-b5 beta on an [ESP32-S3-Tiny](https://www.waveshare.com/wiki/ESP32-S3-Tiny) using any tool

### Expected Behavior

WLED-AP appears after a board reset/power on

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0-b5

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-09-26:
Not a bug of WLED on its own. The S3 is problematic as far as different configurations of SPI RAM and flash memory go.

You will need to compile yourself with correct board selected in PIO and use PIO to flash. Once you have it working you can submit a PR with compile options for your particular board.

See discussion here: https://github.com/Aircoookie/WLED-WebInstaller/issues/22

closed 2024-09-26 by blazoncek

me-dex, 2024-09-26:
Am I wrong to assume that this is only a WLED issue and not a device level issue since Tasmota seems to work without issue? Seems like in the linked post they also point out Tasmota flashes without issue as well