wled/WLED #4162: is it possible to modify the code and make it compatible with raspberry pi pico w. And probably add pio (programmable input output)

Night-Coder2, 2024-09-26:
so i recently got my hands on raspberry pi pico. and used its pio function to manipulate the led behaviour without stressing the processor too much. is it possible to modify the code to be compatible with a raspberry pi pico?

DedeHai, 2024-09-26:
short answer: No.
long answer: maybe, if you have a few people with a few hundred hours of time.

closed 2024-09-26 by DedeHai

blazoncek, 2024-09-26:
Duplicate of #3174

Night-Coder2, 2024-09-26:
i mean the developers might have spent hours on the original code. is it possible to spend a few more to add rpipicow support. its way more efficient and faster because of pio

DedeHai, 2024-09-26:
> i mean the developers might have spent hours on the original code. is it possible to spend a few more to add rpipicow support.
try thousands of hours. and no.

softhack007, 2024-09-26:
As all other maintainers already explained, there are not plans to port WLED to anything not based on espressif controllers. Actually porting to rp pico would be a complete rewrite.

> its way more efficient and faster because of pio

We have RMT and parallel I2S driving on esp32. This is already efficient (up to 14 outputs in parallel) and fast (ws2812 runs at 800000kbit, the [NeoPixelBus](https://github.com/Makuna/NeoPixelBus) driver can deliver leds at full speed).

If you are looking for something more portable, then the [WS2812FX](https://github.com/kitesurfer1404/WS2812FX)  library might be interesting to you.