wled/WLED #4166: [dev problem] The `punycode` module is deprecated

softhack007, 2024-09-29:
Since ~4 weeks, the UI build (npm run build) throws out a warning

>  (node:27944) [DEP0040] DeprecationWarning: The `punycode` module is deprecated. Please use a userland alternative instead.
  #(Use `node --trace-deprecation ...` to show where the warning was created)

I don't know where this tool gets used, or what is the "userland alternative" for punycode. But it looks like the clock is ticking and we need an alternative solution.

blazoncek, 2024-09-30:
I am not sure, but I think npm is mainly used for building include files from html.
Perhaps @w00000dy may know more.

w00000dy, 2024-10-01:
I'll have a look at it later.

w00000dy, 2024-10-01:
@softhack007 What NodeJS and npm version do you use? (`node -v`, `npm -v`)
I could not reproduce the warning that `npm run build` should throw. @blazoncek Do you see this warning?
Are we talking about `0_15` branch?

w00000dy, 2024-10-01:
And the warnings at `npm install` are from the [inliner](https://www.npmjs.com/package/inliner) package. It has not been updated in 7 years.
To fix this, we would have to find an alternative to inliner.

softhack007, 2024-10-01:
> Are we talking about 0_15 branch?

yes, I get these messages on the 0_15 branch

> @softhack007 What NodeJS and npm version do you use? (`node -v`, `npm -v`)

node v22.7.0
npm 10.8.2

w00000dy, 2024-10-01:
I used node v20.17. I will test it again with node v22.

w00000dy, 2024-10-01:
Okay, now I also get the warning with node v22.
This warning comes from the [inliner](https://www.npmjs.com/package/inliner) package.
I think we need to look for an alternative here.

w00000dy, 2024-10-01:
After a short search I found [web-resource-inliner](https://www.npmjs.com/package/web-resource-inliner). At first glance it looks like a good alternative, but I'll have to take a closer look when I have more time.

blazoncek, 2024-10-02:
I am using v20 and I do not see any warnings during `npm run build` I avoid `npm install` but rather use `npm ci`

Sojourneer, 2024-11-03:
I am also having a problem with the UI build for 0_15 (I am trying to test the new translation functionality).
npm run build => Error: Cannot find module 'inliner'

And I just installed 'inline', and successfully invoked it from the command line.
Node.js v22.11.0

Can I have a workaround please.

w00000dy, 2025-03-10:
PR #4266 fixes this.

PR #4266 (2025-03-11): Migrate from inliner to web-resource-inliner
This migrates from the old [inliner](https://www.npmjs.com/package/inliner) package to the [web-resource-inliner](https://www.npmjs.com/package/web-resource-inliner) package.

Now we don't get this warning anymore:
```
(node:29032) [DEP0040] DeprecationWarning: The `punycode` module is deprecated. Please use a userland alternative instead.
(Use `node --trace-deprecation ...` to show where the warning was created)
```

If you are wondering why the flash size increased, it is because the old inline package did not respect [`/*! these comments */ ` ](h [... cut]