wled/WLED #4168: Can't add more than 2 outputs and save

RichyRicho1, 2024-09-30:
### What happened?

When adding the 3rd output, pressing save does nothing. 
Then editing the 1st or 2nd output doesn't allow saving either

### To Reproduce Bug

Add a 2nd output, press save, then try and add a 3rd output and try to save

### Expected Behavior

Should add a 3rd output and save

### Install Method

Binary from WLED.me

### What version of WLED?

WLED Beta Release 0.15.0-b5

### Which microcontroller/board are you seeing the problem on?

ESP32-S2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netmindz, 2024-10-01:
Please also include the web developer console output too. I've seen issues where you can't save due to validation errors only visible in the console

suromark, 2024-10-05:
Hello, I might be able to provide some extra input on the issue "no saving on the preferences" since it has affected my setup as well.
It seems that parts of the HTML form and the javascript do not match (possibly a cache issue?). I've had console reports (in the browser) of "LV4 not found" or errors from UI() function.
Apparently, once multiple strands are configured, the form is not populated correctly. I'm using eight output pins, and the save function would not react if I disable the auto power limiter. I've managed to corrupt the preset completely just by re-enabling the limiter, entering a big number (9000) and save.
Other issues were: all LED pins but the first on the page were set to Custom : 0 mA and then saving failed with a very discrete error message far down on the page, or the last pin not producing usable output but just flickering random colors.

blazoncek, 2024-10-05:
Please provide browser console log.

suromark, 2024-10-05:
Here's the current browser log, plus a screenshot of the LED interface.

leds:1 An invalid form control with name='LA1' is not focusable. <input name=​"LA1" type=​"number" min=​"1" max=​"255" oninput=​"UI()​">​
leds:1 An invalid form control with name='LA2' is not focusable. <input name=​"LA2" type=​"number" min=​"1" max=​"255" oninput=​"UI()​">​
leds:1 An invalid form control with name='LA3' is not focusable. <input name=​"LA3" type=​"number" min=​"1" max=​"255" oninput=​"UI()​">​
leds:1 An invalid form control with name='LA4' is not focusable. <input name=​"LA4" type=​"number" min=​"1" max=​"255" oninput=​"UI()​">​
leds:1 An invalid form control with name='LA5' is not focusable. <input name=​"LA5" type=​"number" min=​"1" max=​"255" oninput=​"UI()​">​
leds:1 An invalid form control with name='LA6' is not focusable. <input name=​"LA6" type=​"number" min=​"1" max=​"255" oninput=​"UI()​">​

[image]

Basically, once I switch off the automatic brightness limiter, the form becomes unusable. Once i switch it back on, all except the first output default to "mA/LED: Custom 0 mA".

After I enter some value ("seed/Fairy 15 mA") and try to save, there's no more response from the webserver, and the ESP32 resets. Reloading the page after that only shows that none of the changes were stored in the ESP.

So far I've found I need to do a factory reset and reconfigure everything to have a chance at trying again.

I've seen that sometimes it works if I enter the GPIO pin numbers backwards, i.e. instead of doing channels 1-8 as 16, 17, 18, 19, 21, 22, 23, 25, it may work if I do 25, 23, 22, 21, 19, 18, 17, 16. That's not guaranteed though.

Also, instead of saving "GPIO25", I've seen it shows "26" once I reload the page. This might also explain why (see video) the ring light on the side (which is tied to GPIO25) does not go to the default orange output but seems to catch some test signal?

https://github.com/user-attachments/assets/8a86afec-37c2-4bc7-864f-b65d03090c07

softhack007, 2024-10-05:
> _Which microcontroller/board are you seeing the problem on_?
> ESP32-S2

So you are not using ESP32-S2, right? Because-S2 does not allow for more than 4 outputs.

Technicially speaking, it does not make much sense to have 8 outputs where each one only controls 12 leds. 

You could simply use just one output pin, and chain all ws2812b parts together. With this setup you can still create a segment for each "element" in your chain, and have like 8 different effects running in parallel (one for each segment). This will also reduce the CPU load, improve stability and help avoid flickering.

That said, it looks like there is a problem in the led settings page when ABL is disabled.

suromark, 2024-10-05:
Oh, sorry, I forgot. Yes, it's a "classic" ESP32 WROOM-32U devkit v4 module. Why I'm using 8 separate channels is that each of the fans only has an input, I can't chain them into one long strand. I'm using OpenRGB to push E1.31 data so each fan has its own "gauge reading"; the rear one shows network load, the front ones do CPU load, GPU load and CPU core temperature. Funny enough, I managed to get the same version running on another setup, without the odd flickering. I just can't seem to replicate that any more.

suromark, 2024-10-06:
I can add some more details about the flickering / output issue; it happens when I set the the last segments to different lengths. The working configuration has all 8 output channels set to 12 LEDs each, which causes no issues. Once I set the last channels to e.g. 32, they just flicker random colors, as if the output is taken from some random buffer or some buffers got overlaid. The offsets show in the configuration seem OK though.

I guess I should better make this a separate issue?

blazoncek, 2024-10-06:
I think I found the issue.

blazoncek, 2024-10-06:
> Basically, once I switch off the automatic brightness limiter, the form becomes unusable. Once i switch it back on, all except the first output default to "mA/LED: Custom 0 mA".

That's the unfortunate side effect. Per port ABL is a recent addition and may have bugs that are revealed with unusual configuration. It is also quite complex.

I will ask if you can compile yourself using debug flags (-D WLED_DEBUG) and provide serial output traces when buses are saved and/or loaded. LED configuration is saved once all buses are correctly initialised which is why you may get different settings values once you pres save and then reopen settings. If some buses do not initialise correctly settings will show that.

suromark, 2024-10-06:
I tried to compile on Windows 11 Pro with a pretty basic install of Platformio / Visual Studio Code. I checked out the b6 branch but I get a couple of errors with "Cannot find module 'inliner'" and "wled00/wled_server.cpp:3:21: fatal error: html_ui.h: No such file or directory". I guess my setup isn't right, but I'm doing some arduino on the side only and I don't really know what to do at this point :-/

softhack007, 2024-10-06:
> errors with "Cannot find module 'inliner'" and "wled00/wled_server.cpp:3:21: fatal error: html_ui.h: No such file or directory"

Looks like you are missing "npm" (node.js). After installing node.js, open a terminal window in VSCode and run "npm install". This is necessary once to install the tools for compiling UI files.

* https://kno.wled.ge/advanced/compiling-wled/ (preparations, step2 and step5)
* https://github.com/Aircoookie/WLED-Docs/issues/161 (selecting a firmware build, build&upload from VSCode)

blazoncek, 2024-10-07:
> I checked out the b6 branch

b6 is not the correct branch. use regular 0_15

suromark, 2024-10-07:
OK, I think I got it. I had already installed node.js 20 for windows but that failed (i guess?) during the extra install of chocolatey. Running "npm install" from VSC terminal worked, I got b6 to compile with only a couple of minor warnings. Switched the branch to 0_15 which also built OK and flashed with no issues. I've not managed to get it to build in debug mode though... I added -D DEBUG to the build_flags in common_mode but did not see any change in the serial monitor output when I did a config save on the web interface.

I can however confirm that with branch 0_15 now I can set the LED channels to different lengths without the lights going manic :-D so thank you very much!

Example: 7 x 12 LED fans + 1 x 50 LEDs 2.7mm strip.

[image]

https://github.com/user-attachments/assets/1be190a3-bf1a-4ad5-8ba5-570f2e31c63e

suromark, 2024-10-08:
Sadly it's not quite fixed yet; after trying various settings I once again have flicker issues as soon as I define one segment with a different number of LEDs as the others. I've attached a video where I start with all channels set to 12 LEDs (which works perfectly), then change the center one to 24 LEDs length (at about 3 seconds into the video) which causes it to flicker as if its buffer is colliding with its neighbor? I can't change this back using the HTML form, but I can use "security & updates" to upload a backup JSON cfg to have it work again.

https://github.com/user-attachments/assets/699647ed-ac71-4cf9-9060-58e0081d2ac9

blazoncek, 2024-10-08:
That's entirely different issue than OP.
You may want to explore segment settings and please use WLED [forum](https://wled.discourse.group) or [Discord ](https://discord.gg/FMvz8nzmat)for help and support.

closed 2024-10-08 by blazoncek