wled/WLED #4169: Adding support for new LED (HD108)

mkovero, 2024-10-02:
### What happened?

Hi, as I did simple support for HD108 led in NeoPixelBus (https://github.com/Makuna/NeoPixelBus/pull/838) I thought doing so on WLED as well to help out my personal projects a bit.
Naively I thought that simply 
a) cloning B_HS_DOT_3 / B_SS_DOT_3 magic as an example in bus_wrapper.h with new HD108 methods and 
b) incrementing I_HS_DOT_3 / I_SS_DOT_3 id's by one after last known definition,
`#define I_HS_HD108_3 63`
`#define I_SS_HD108_3 64`
c) adding new type in const.h
`#define TYPE_HD108               55`
d) adding new option in settings_leds.htm (and rebuilding the html headers afterwards)
`<option value="55">HD108</option>\`

would do the trick but unfortunately I only get crash & reboot after selecting HD108 from the settings page.
However, if I just replace the methods used by APA102 in bus_wrapper.h I'm able to use LEDs fine.
So I'm not that far am I? Something fancy happening with id's I should be aware of?)
Currently I do not have board I could debug but will get there if nothing else helps)

Thanks for your time)

### To Reproduce Bug

Try adding support for new LED strip.

### Expected Behavior

To be able to do so.

### Install Method

Self-Compiled

### What version of WLED?

latest master

### Which microcontroller/board are you seeing the problem on?

ESP32, ESP32-S2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netmindz, 2024-10-02:
Hi @mkovero 

Thanks for your interest in adding support for extra LED types. We have recently done some work to make this a bit easier, so please create your branch from the 0_15 branch. You will no longer need to edit the html at all.

I'm not sure about the other changes required as I'm not familiar with this LED type and what is required to use it with NPB

Krayn, 2024-11-05:
> ### What happened?
> Hi, as I did simple support for HD108 led in NeoPixelBus ([Makuna/NeoPixelBus#838](https://github.com/Makuna/NeoPixelBus/pull/838)) I thought doing so on WLED as well to help out my personal projects a bit. Naively I thought that simply 
a) cloning B_HS_DOT_3 / B_SS_DOT_3 magic as an example in bus_wrapper.h with new HD108 methods and 
b) incrementing I_HS_DOT_3 / I_SS_DOT_3 id's by one after last known definition, 
`#define I_HS_HD108_3 63` 
`#define I_SS_HD108_3 64` 

It Seems the IDs you used are already in use:
`#define I_32_RN_FW6_5 63`
`#define I_32_I0_FW6_5 64`

> c) adding new type in const.h 
`#define TYPE_HD108 55` 
d) adding new option in settings_leds.htm (and rebuilding the html headers afterwards) 
`<option value="55">HD108</option>\`
> would do the trick but unfortunately I only get crash & reboot after selecting HD108 from the settings page. However, if I just replace the methods used by APA102 in bus_wrapper.h I'm able to use [...]

@mkovero Could you share your changes to turn the APA102 into HD108? I am currently trying to get my HD108 strip working without success (neither with a new LED type or reusing the APA102 definitions)

Mpxxer, 2024-11-08:
@mkovero any news on the HD108 implementation? this would be really great to have support for hd108 in WLED!
@netmindz or is it a topic for WLED? i think (but i am not coding) hd108 is a 16 bit led and excellent for dimming for 0-255. in Pixelblaze its supportet...would be great if @Aircoookie could follow:-)

DedeHai, 2024-11-08:
besides adding basic hardware support:
WLED is all 8bit color based, so you also need to add a 'correction factor' like it is done for white channel PWM (CIE lookup table) or other color correction functions so make use of the increased color depth.

Mpxxer, 2024-11-08:
@DedeHai thank you very much for your comment! do you know any efforts in this matter in actual or future releases?

DedeHai, 2024-11-08:
we talked about adding alpha channel, but going 16bit for each color is just not feasible, RAM and flash memory are already at the limit. But converting the 24bit RGB values to 48bit with a correction curve is pferfectly doable, it just requires someone with the correct testing gear to write that code and test it across all systems.

mkovero, 2024-11-11:
> It seems the IDs you used are already in use: `#define I_32_RN_FW6_5 63` `#define I_32_I0_FW6_5 64` (actually i just rechecked..they are only already in use on 0_15 branch)

Ah! Glad this was so!)

> @mkovero Could you share your changes to turn the APA102 into HD108? I am currently trying to get my HD108 strip working without success (neither with a new LED type or reusing the APA102 definitions)

I certainly can within this week or the next, will let you know how it is!)

Mpxxer, 2024-11-26:
@mkovero any news?

Mpxxer, 2024-12-19:
in wled 15 12 bit brightness control added but not for digital led :-(
How about HD108 implementation @mkovero any news? Can I hire someone who can do this?

Blaster1920, 2025-02-05:
This is what I'm doing at the moment on my standalone implementation of the HD108 for esp32-c6 . I'm REALLY looking forward to help devs in wled and fastled to sort them out.

Notice some quirks I'm having to do at the moment, and sorry for the poor coding.
Eg:  

- `sendLEDFrame(1, 0, 0, 0, 0, 0); // Turn off all LEDs` has to have at least a 1 or the command won't be seen by the strip. 

-  `// Automatically set gain to max if grayscale is 0, this avoids LEDs crashing` same applies here

```

// Initialize SPI
void HD108::begin() {
    spi_bus_config_t buscfg = {
        .mosi_io_num = _dataPin,
        .miso_io_num = -1,
        .sclk_io_num = _clockPin,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .max_transfer_sz = 64 * _numLEDs + 16 // Max transfer size
    };

    esp_err_t ret = spi_bus_initialize(_spiHost, &buscfg, SPI_DMA_CH_AUTO);
    assert(ret == ESP_OK);

    spi_device_interface_config_t devcfg = {
        .mode = 3,                   // SPI mode 3
        .clock_speed_hz = 15000000,  // 20 MHz clock speed
        .spics_io_num = -1,          // No CS pin
        .queue_size = 1,             // Transaction queue size
    };

    ret = spi_bus_add_device(_spiHost, &devcfg, &_spi);
    assert(ret == ESP_OK);

        sendStartFrame();
    for (int i = 0; i < _numLEDs; i++) {
        sendLEDFrame(1, 0, 0, 0, 0, 0); // Turn off all LEDs
    }
    sendEndFrame();
}

// Send start frame
void HD108::sendStartFrame() {
    uint8_t startFrame[4] = {0x00, 0x00, 0x00, 0x00}; // 32-bit start frame
    spi_transaction_t t = {
        .length = 32, // 32 bits = 4 bytes
        .tx_buffer = startFrame,
    };
    spi_device_transmit(_spi, &t);
}

// Send LED frame
void HD108::sendLEDFrame(uint8_t redGain, uint8_t greenGain, uint8_t blueGain,
                         uint16_t red, uint16_t green, uint16_t blue) {
    uint8_t frame[8];

    // Automatically set gain to max if grayscale is 0, this avoids LEDs crashing
    if (red == 0) redGain = 1;
    if (green == 0) greenGain = 1;
    if (blue == 0) blueGain = 1;

    // Start Code + Current Gain Adjustment (15 bits)
    frame[0] = 0x80 | ((redGain & 0x1F) << 2) | ((greenGain >> 3) & 0x03); // Start code + Red MSB + Green MSB
    frame[1] = ((greenGain & 0x07) << 5) | (blueGain & 0x1F);              // Green LSB + Blue 5 bits

    // Encode RGB 16-bit values
    frame[2] = (red >> 8) & 0xFF;   // High byte of Red
    frame[3] = red & 0xFF;          // Low byte of Red
    frame[4] = (green >> 8) & 0xFF; // High byte of Green
    frame[5] = green & 0xFF;        // Low byte of Green
    frame[6] = (blue >> 8) & 0xFF;  // High byte of Blue
    frame[7] = blue & 0xFF;         // Low byte of Blue

    // Transmit the frame via SPI
    spi_transaction_t t = {
        .length = 64, // 64 bits = 8 bytes
        .tx_buffer = frame,
    };
    spi_device_transmit(_spi, &t);
}

// Send end frame
void HD108::sendEndFrame() {
    uint8_t endFrame[4] = {0xFF, 0xFF, 0xFF, 0xFF}; // At least 1 bit per LED
    spi_transaction_t t = {
        .length = (_numLEDs / 16) * 8, // End frame size
        .tx_buffer = endFrame,
    };
    spi_device_transmit(_spi, &t);
}

// Show function
void HD108::show(uint8_t* redGain, uint8_t* greenGain, uint8_t* blueGain,
                 uint16_t* red, uint16_t* green, uint16_t* blue) {
    sendStartFrame();
    for (int i = 0; i < _numLEDs; i++) {
        sendLEDFrame(
            redGain[i],    // Red gain
            greenGain[i],  // Green gain
            blueGain[i],   // Blue gain
            red[i],        // Red intensity
            green[i],      // Green intensity
            blue[i]        // Blue intensity
        );
    }
    sendEndFrame();
}

void HD108::clear() {
    sendStartFrame();
    for (int i = 0; i < _numLEDs; i++) {
        sendLEDFrame(0, 0, 0, 0, 0, 0); // Turn off all LEDs: gain and grayscale set to 0
    }
    sendEndFrame();
}

void HD108::showColour(const char* colorName, uint16_t ledIndex) {
    uint8_t redGain, greenGain, blueGain;
    uint16_t red, green, blue;

    getColourParameters(colorName, redGain, greenGain, blueGain, red, green, blue);

    sendStartFrame();
    for (int i = 0; i < _numLEDs; i++) {
        if (i == ledIndex) {
            sendLEDFrame(redGain, greenGain, blueGain, red, green, blue);
        } else {
            sendLEDFrame(0, 0, 0, 0, 0, 0); // Turn off other LEDs
        }
    }
    sendEndFrame();
}
```

sjonkeesse, 2025-05-16:
Any updates on this? I’m looking forward to have a higher brightness resolution with my HD108 ledstrip with WLED.

netmindz, 2025-05-28:
Not currently is my understanding @sjonkeesse