wled/WLED #4170: WLED does not remember the latest wifi it has been connected to

heisenberg2980, 2024-10-02:
### What happened?

I originally connected my WLED to a wifi network, I have now changed the name of the wifi to a different name, and every time I restart my Home Assistant or there is an issue, WLED disconnects from the network. After connecting to the WLED access point and enter the wifi menu I can see the wifi stored there is the original wifi and not the one I later setup, and this happens again and again

### To Reproduce Bug

Setup the WLED controller with one wifi and change it to a different one. After that restart Home Assistant and the WLED will disconnect from the network

### Expected Behavior

WLED should store the latest wifi it has been connected to

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.4 (build 2405180)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-10-05:
Try to restore factory defaults and retry.
Be sure to backup configuration and presets and any other file uploaded to file system.

heisenberg2980, 2024-10-05:
@blazoncek I just tried that, and the previous wifi settings must be in the configuration backup file, because as soon as I restored the configuration from the backup the same issue happened again (it disconnected from the wifi, and I had to connect to the WLED access point and the old wifi name was in wifi settings).

blazoncek, 2024-10-05:
You can remove `"nw":{"ins":[...]}` from the config file.
Use a tool like jsonformatter.org to beautify and then minimize JSON file.

heisenberg2980, 2024-10-05:
I will try, but shouldn´t WLED save the latest wifi network it has been connected to instead of the original wifi network that was set up initially on the device?

blazoncek, 2024-10-05:
It does. At least on my devices.

heisenberg2980, 2024-10-05:
Have you ever tried to change your WLED devices to a different wifi name to check this behaviour? To be honest this issue is not easy to discover, as only people who have changed their wifi name will experience this. An easy way to test it would be to create a new wifi (some routers and access points allow you to do this), connect the WLED device to the new wifi and wait some days to see if the device is still connected to the new wifi after restarting Home Assistant and the device itself multiple times.

blazoncek, 2024-10-06:
You are assuming too much. 😄

dosipod, 2024-10-13:
@heisenberg2980 
I do this a lot as I  travel so I would create a new wifi SSID all the time and wled will save it .For HA thing which I did not go into and might be the source of your issue you can remove and add the wled device once it is connected to the new SSID . And please do the testing on latest source if possible   

Just look under edit/cfg.json and you will see which ssid is saved so no magic there  
[image]

heisenberg2980, 2024-10-27:
@dosipod this issue is really bothering me. I agree with you that HA might be the source of my issue, but I already removed the WLED from HA and added it again, and the issue reappeared. 

I am running the latest stable version of both HA and WLED (0.14.4), and I have created a backup of the configuration in the WLED where I can see the SSID is correct, and then after some time I have done a new backup and the SSID has been changed to the old one.

I haven´t even been able to find the pattern of when that change happens, as I have restarted both WLED and HA and the SSID is still correct, and then a couple of days later I enter the settings and I can see it´s already been changed to the old one (which is weird, as the WLED is still connected to the correct SSID, but the settings point to the old SSID, which means as soon as WLED restarts for any reason it's unable to connect)

dosipod, 2024-10-28:
@heisenberg2980 I think a simple test that was mentioned in the start of the post would be to reset the unit and then see the behavior afterword since you already have a backup  . But it seem a bit odd  behavior which I could not replicate and I have multiple routers and do switch SSID all the time for testing

heisenberg2980, 2024-10-28:
@dosipod yes I did that test as well, I did a Factory reset and then restored the backup (checking the backup file and ensuring the correct SSID was there). After restoring the backup I could see in the WLED settings that the right SSID was set (I checked again after a few hours and it was still correct), and still, after a day or two I opened the settings again and the old SSID was there, it´s like a ghost that keeps coming back to scare me 😄

heisenberg2980, 2024-10-28:
@dosipod I have a new update and it´s even more strange. I removed the WLED from Home Assistant, then I set the correct SSID in the WLED settings and I checked the SSID every hour, and the correct SSID was there during the first 5 hours, but when I checked again after 5 hours the SSID had been changed to the old one again! So it doesn´t seem HA is involved in this issue.

dosipod, 2024-10-28:
@heisenberg2980 So when you backup your cfg,json and view it  do you see the old  SSID  in the file . If the answer is no then must be something external  updating that or it is cached somewhere .
At the point I would personally test by doing a factory reset of the unit and not restore  any backups but set it up as new just to get to the bottom of this , if you also provide your cfg.json I could load that on a test unit and keep it for few days but if the old ssid is not in the file then there is no way I would get your old SSID

blazoncek, 2024-10-28:
Erase flash, flash b6, configure at a friend (or some other SSID), verify.

heisenberg2980, 2024-10-28:
@dosipod I found the issue, and I am ashamed, because as usual in these weird issues, the problem was with the user (me) 😄

I found an automation I created ages ago in Home Assistant that would send an http request to the WLED to change the settings according to certain conditions that happened rarely, and this request included the old SSID.

So the problem is now solved, and hopefully this issue will help anyone in the future having a similar issue (at least I want to think that to feel better...)

closed 2024-10-28 by heisenberg2980

blazoncek, 2024-10-28:
And you wasted 6 hours of my life, trying to figure out what's causing it. 🤦‍♂️ 
Oh, well...

dosipod, 2024-10-28:
@heisenberg2980  At least now you know it was not the ghost  . Wish you luck