wled/WLED #4171: Not able to compile WLED with Visual Studio Code

TPPTim, 2024-10-02:
### What happened?

Hello,
I am not able to compile the WLED project using Visual Studio Code. I followed this guide: https://kno.wled.ge/advanced/compiling-wled/

When I hit the **build** check button on the bottom PIO bar, only this line popps up in the search fild.
[image]

Nothing else happened. The Output shows:
[image]

I did not change any thing else in the WLED project folder. Can you please help me what is wrong?

### To Reproduce Bug

Install Git
Install Visual Studio Code
Install Platform IO Plugin
Open WLED project folder

### Expected Behavior

compiling the project

### Install Method

Self-Compiled

### What version of WLED?

0.14.4

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netmindz, 2024-10-02:
You should really be selecting which build env to use. Please also include a screenshot of what is shown when you select platformio on the right

TPPTim, 2024-10-02:
When I hit the build check button still nothing happened.
[image]

When I click on build at the left side in the menue, then it starts to compile:
[image]

Is this behavior ok?

blazoncek, 2024-10-02:
That is a known issue with PIO since update sometime in late 2022.

TPPTim, 2024-10-02:
OK thank you very much! What is the common solution for this problem.
Is there an work around or should I use the Build option from the left sided Menue bar instead in future?

TPPTim, 2024-10-04:
Thank you very much! I have tried this, but I get an error that I cannot explain.
[image]

For this I created the **platformio_override.ini** in the main Folder and added some **-D Pindefines**:
[image]

blazoncek, 2024-10-04:
Github issues are not intended to educate users.
Please use WLED [forum](https://wled.discourse.group) or [Discord](https://discord.gg/FMvz8nzmat) for help and support questions.

closed 2024-10-04 by blazoncek

TPPTim, 2024-10-04:
OK - thank you!
I created a topic here: [](https://wled.discourse.group/t/compile-wled-with-visual-studio-code-platformio-issue/12860)