wled/WLED #4177: Stop turning the lights/relay on when changing the brightness

Andro-Marian, 2024-10-03:
**Is your feature request related to a problem? 
The leds are turning on when changing the brightness A=x...z

**Describe the solution you'd like**
A way to stopthis to happen.

**Describe alternatives you've considered**
Add a way to set "on":null to keep the current state.

DedeHai, 2024-10-03:
can you give a concrete example?
why would you want to change brightness on a led that is off and why should that not turn it on?

softhack007, 2024-10-03:
> The leds are turning on when changing the brightness A=x...z
> 
>  A way to stop this to happen if there is not a "on":true

Hi, as @DedeHai says, it would help to understand the use case that's behind your request.

As of today, WLED does not have any "on" or "off" state (even when there is a button to switch light on/off). Internally WLED only uses the global brightness value to control the relay:

* brightness = 0 ==> relay off (leds not powered)
* brightness  > 0 ==> relay on (leds powered)

Andro-Marian, 2024-10-04:
Using with time presets to set the brightness.
And when turning the lights off, the preset turns them on.

Maybe using a light senson may also break this?

blazoncek, 2024-10-04:
You can use JSON API to change brightness and have LEDs remain off.
I.e.: `{"on":false,"bri":100}`

closed 2024-10-04 by blazoncek

DedeHai, 2024-10-04:
I think the scenario provided is a valid point if I understand correctly:
- scheduled brightness, say dim to bri=50 at 10PM
- if lights are switched off before that time, they should not turn back on
is there a way to do this already?
in code, this could be solved by adding checks: there is the variable offmode (or something like that). I am pretty sure it would brake existing behaviour for some users though as they do expect changing brightness to turn the lights on.

blazoncek, 2024-10-04:
Changing WLED state to on when brightness changes >0 is a default behaviour since pre 0.10 version.
This is especially true with HTTP API.

JSON API was enhanced to allow changing brightness while keeping WLED state off (example provided above). It will not be ported to HTTP API.

DedeHai, 2024-10-04:
if this is something many would appreciate, it could be a setting "do not turn on upon brightness change".

Andro-Marian, 2024-10-04:
> I think the scenario provided is a valid point if I understand correctly:
>
>     * scheduled brightness, say dim to bri=50 at 10PM
>
>     * if lights are switched off before that time, they should not [...]

Yes, is right. If I sheduled a brightness change and the lights are off  should stay off.

Not sure why this change was made by the brightness.
If I understand, when the power buttons is turning off the current brightness is saved already to another variable.

Why you don't just separate first by second.
And add a options on LED Settings to add this current behaviour.

Anyway I don't understand this behaviour anyway.

DedeHai, 2024-10-04:
> Why you don't just separate first by second.
> And add a options on LED Settings to add this current behaviour.

you are invited to provide a PR if this is important to you. Or find more supporters for this.

I also do not follow the logic but it is a legacy feature that 'has always been this way' so many setups may depend on it working this way.

Andro-Marian, 2024-10-04:
I don't know how it's coded/implemented. I just assumed is this way.

I also find more posts about this on other places.

Andro-Marian, 2024-10-08:
Who need this fix edit the json.cpp -> deserializeState:

```
  bool onBefore = bri;
  bool on = root["on"] | (bri > 0);

  if (on) {
    getVal(root["bri"], &bri);
  } else {
    getVal(root["bri"], &briLast);
  }

  if (!on != !bri) toggleOnOff();
```

Andro-Marian, 2025-03-02:
Found another reason for this when using the LedFx.
If I change the brightness when on Live the device becomes ON and when turning the LedFx off the device remaining ON.