wled/WLED #4178: D1 mini - device becomes slow/unresponsive *only when LEDs switched on* - even with no LED connected.

goodlucknow, 2024-10-03:
### What happened?

I'm encountering an issue on multiple D1 mini devices - I thought I was having extremely bad luck with a faulty batch.
Issue occurs only from 14.1 upwards, and not on all D1 mini devices but several in my house.

UI is blazing fast and responsive while that "power" toggle is set to off. As soon as I toggle on the UI lags & loses connection, will not reload etc. I can sometimes turn off via API call, or reboot the device(my default boot preset is with LEDs off, how I discovered this behaviour).

This occurs on a bare board D1 mini powered over USB with nothing else connected, no LEDs.

I have just discovered the latest beta "compat" version does not have the issue.
I see the announcement mentions this is to do with wifi so could explain it, but I thought the behaviour strange enough to warrant opening a bug report in case it could be of use.

### To Reproduce Bug

D1 mini, environment ESP8266 3.12. (2)
Any WLED version from 14.1 upwards.

### Expected Behavior

Normal connectivity

### Install Method

Binary from WLED.me

### What version of WLED?

>=14.1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

willmmiles, 2024-10-04:
Duplicate - this issue is being tracked here: #3685

closed 2024-10-04 by willmmiles

goodlucknow, 2024-10-04:
Thanks for the link @willmmiles , I didn't realise it was the same issue as I hadn't observed rebooting in my case and I thought the power toggling causing the issue even with no LEDs connected unusual. It is rock solid stable if left switched "off".
I'll have a read through there and hopefully learn something. Perhaps time for me to start compiling instead of using the release binaries.