wled/WLED #4179: Pacifica effect on 0.15.0-b5 shows a flicker that wasn't present in 0.14.4

KEANO89, 2024-10-05:
### What happened?

0.15.0-b5 (didn't check other betas yet) produces a sort of irregular but frequent flicker when the pacifica effect is selected (regardless of palette or brightness or level shifter) but other effects were not affected in my quick testing. Reverting back to 0.14.4 fixes the issue without hardware changes.

### To Reproduce Bug

Use 0.15.0 beta and select the effect "Pacifica"

Hardware: esp32 (powered from the same 24v source but obv stepped down to 5v), bft rgbw 24v strip, level shifter from 3.3 to 5 on the data line

### Expected Behavior

Same as in 0.14.4 or earlier versions

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0-b5

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-10-05:
Can you post a video depicting the issue? I am unable to reproduce.

KEANO89, 2024-10-05:
[Courtesy of ALDIY](https://discord.com/channels/473448917040758787/473449548279185408/1292056657827729450). 
I'm going to record one of my setup later today once I've untangled my testing ~mess~ setup

DedeHai, 2024-10-05:
I do see it on my C3, there is periodic brightness jumps. Bug was introduced in 68a7282b2765f4c43c838e7554a93a95f4385406

DedeHai, 2024-10-05:
reverting https://github.com/Aircoookie/WLED/blob/ba636b17a0a705baaca3f3c667d3396c41294ab9/wled00/FX.cpp#L4034
back to uint16_t solves it.

closed 2024-10-05 by blazoncek