wled/WLED #4187: ESP32 won't connect to WiFi when SSID is using both 2.4 and 5 Ghz

jf-navica, 2024-10-08:
### What happened?

ESP-WROOM-32 won't connect to WiFi when SSID is the same on both 2.4 and 5 Ghz.  If the SSID is only on 2.4 Ghz, it connects.  It appears to be related to the way Synology handles band steering.  BUT...  Bug is not present on ESP8266 and using ESPHOME on same ESP-WROOM-32 works fine both using Arduino and ESP-IDF frameworks.

### To Reproduce Bug

erase flash
write_flash 0x0 ./esp32_bootloader_v4.bin
write_flash 0x10000 ./WLED_0.15.0-b5_esp32_debug.bin (or any other versions down to 13.3)
Connect to AP, configure WiFi

### Expected Behavior

It should connect to WiFi

### Install Method

Self-Compiled

### What version of WLED?

0.15.0-b5

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
0x1 (POWERON_RESET),boot:0x13 (SLocal time: 00:02
---DEBUG INFO---
Runtime: 120063
Unix time: 120,063
Free heap: 179968
TX power: 62/78
Wifi state: 6
ESP-NOW state: 0
State time: 30000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 1636
Loop time[ms]: 0/2
UM time[ms]: 0/1
Strip time[ms]:0/1
Segments: 1 -> 68/0B
  Seg: 30,1 [A=1, 2D=0, RGB=1, W=0, CCT=0]
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

---WLED 0.15.0-b5 2409170 INIT---

esp32 v3.3.6-16-gcc5440f6a2
arduino-esp32 v1.0.x

CPU:   ESP32-D0WDQ5 rev.3, 2 core(s), 240 MHz.
FLASH: 4 MB, Mode 3 (DOUT), speed 40 MHz.
heap 302988
[... 382 lines cut ...]
Free heap: 181636
TX power: 72/78
Wifi state: 6
ESP-NOW state: 0
State time: 120064
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 1571
Loop time[ms]: 0/44
UM time[ms]: 0/1
Strip time[ms]:0/29
Segments: 1 -> 68/0B
  Seg: 30,1 [A=1, 2D=0, RGB=1, W=0, CCT=0]
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
```

### Anything else?

Not self-compiled.  Used debug binaries provided by: https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/%40Aircoookie/Dev/0.15.0-b5/Debug_builds

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-10-08:
I have Ubiguiti's Unifi network and everything works with the same SSID on 2.4GHz and 5GHz.
I would suspect something wrong on your particular WiFi settings.

WLED uses stock Arduino/IDF WiFi stack so issue should be reported (and debugged) there.

jf-navica, 2024-10-08:
Can't open a bug there, ESPHOME also uses stock Arduino/IDF and it works.  So what's different between WLED and ESPHOME?

blazoncek, 2024-10-08:
WLED (in 0.15) does:
```c++
...

  WiFi.persistent(false);
  WiFi.onEvent(WiFiEvent);
#if defined(ARDUINO_ARCH_ESP32) && ESP_IDF_VERSION_MAJOR==4
  WiFi.useStaticBuffers(true);    // use preallocated buffers (for speed)
#endif
#ifdef ESP8266
  WiFi.setPhyMode(force802_3g ? WIFI_PHY_MODE_11G : WIFI_PHY_MODE_11N);
#endif
  if (isWiFiConfigured()) {
    showWelcomePage = false;
    WiFi.setAutoReconnect(true);  // use automatic reconnect functionality
    WiFi.mode(WIFI_MODE_STA);     // enable scanning
    findWiFi(true);               // start scanning for available WiFi-s
  } else {
    showWelcomePage = true;
    WiFi.mode(WIFI_MODE_AP);      // WiFi is not configured so we'll most likely open an AP
  }

...

  if (isWiFiConfigured()) {
    WiFi.mode(WIFI_MODE_STA); // engage explicit STA mode
    // determine if using DHCP or static IP address, will also engage STA mode if not already
    if (multiWiFi[selectedWiFi].staticIP != 0U && multiWiFi[selectedWiFi].staticGW != 0U) {
      WiFi.config(multiWiFi[selectedWiFi].staticIP, multiWiFi[selectedWiFi].staticGW, multiWiFi[selectedWiFi].staticSN, dnsAddress);
    } else {
      WiFi.config(IPAddress((uint32_t)0), IPAddress((uint32_t)0), IPAddress((uint32_t)0));
    }

    // convert the "serverDescription" into a valid DNS hostname (alphanumeric)
    char hostname[25];
    prepareHostname(hostname);

#ifdef ARDUINO_ARCH_ESP32
    WiFi.setTxPower(wifi_power_t(txPower));
    WiFi.setSleep(!noWifiSleep);
    WiFi.setHostname(hostname);
#else
    wifi_set_sleep_type((noWifiSleep) ? NONE_SLEEP_T : MODEM_SLEEP_T);
    WiFi.hostname(hostname);
#endif
    WiFi.begin(multiWiFi[selectedWiFi].clientSSID, multiWiFi[selectedWiFi].clientPass); // no harm if called multiple times
    // once WiFi is configured and begin() called, ESP will keep connecting to the specified SSID in the background
    // until connection is established or new configuration is submitted or disconnect() is called
  }

...
```

`WiFi` being instance of generic Arduino `WiFi` class.

softhack007, 2024-10-08:
> won't connect to WiFi when SSID is the same on both 2.4 and 5 Ghz. If the SSID is only on 2.4 Ghz, it connects.

In fact esp32 only has 2.4Ghz radio, so it cannot even "see" the 5Ghz network. 

So your router might have features enabled (fast roaming, 2.4ghz -> 5ghz handover, or whatever) that causes the behaviour you describe. But we can't control that from WLED side. Again, your esp32 does not even "see" the 5Ghz frequencies.

jf-navica, 2024-10-08:
> > won't connect to WiFi when SSID is the same on both 2.4 and 5 Ghz. If the SSID is only on 2.4 Ghz, it connects.
> 
> In fact esp32 only has 2.4Ghz radio, so it does nor even know the 5Ghz network exists.
> 
> Your router might have features enabled (fast roaming, 2.4ghz -> 5ghz handover, or whatever) that causes the behaviour you describe. But we can't control that from WLED side. Again, your esp32 does not even "see" the 5Ghz frequencies.

That I figured out!  I'm just puzzled as to why one works but not the other.   I'll try the V4 firmware and let you know

dosipod, 2024-10-08:
I have setup a router(apple time capsule)   in response to the OP issue in discord  yesterday  with SSID the same on both 2.4 and 5 Ghz and i did not face an issue as the wifi scan page will still pick up the SSID for 2.4

jf-navica, 2024-10-08:
> I have setup a router(apple time capsule) in response to the OP issue in discord yesterday with SSID the same on both 2.4 and 5 Ghz and i did not face an issue as the wifi scan page will still pick up the SSID for 2.4

Same here, they all show up in scan page.  But connection won't work

dosipod, 2024-10-08:
@jf-navica  While it is not a real solution and could be not suited for you , it might be best to get 2nd  router ( i have two TP-Link WR840N ) just for the IoT devices  and connect that to your main router as that really a solution to a lot of issues but otherwise you might have to dig in your main router setting  which is not easy due to disruption .  Please get one in all cases

jf-navica, 2024-10-08:
> > So what's different between WLED and ESPHOME?
> 
> @jf-navica there are lots of differences - most importantly the core libraries used. I'm not sure what ESPHome uses, however i'm relatively sure its esp-idf 4.4.x and arduino-esp32 2.0.y.
> 
> For various reasons (especially flash usage), WLED for esp32 is still on esp-idf 3.5.x and arduino-esp32 1.0.6. Before you ask - no we cannot "simply upgrade" these core versions, unless you (as a user) are OK with getting a firmware that only runs on 8MB flash boards.
> 
> We actually have a "V4" build that uses the new espressif core, but its still experimental because we cannot (yet) fit all WLED features into this build. Good news is, this build can be uploaded to 4MB devices.
> 
> You can go to https://wled-install.github.io/ (alternative web installer) and flash your board with a "V4" firmware. This should help to find out if the newer wifi core makes a difference. Please make a backup of your config & presets because the installer will erase the flash memory.
> 
> ![image](https://private-user-images.githubusercontent.com/91616163/374584392-1dd72b16-7c00-46b0-ac07-5dec4715762c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgzOTcxNTcsIm5iZiI6MTcyODM5Njg1NywicGF0aCI6Ii85MTYxNjE2My8zNzQ1ODQzOTItMWRkNzJiMTYtN2MwMC00NmIwLWFjMDctNWRlYzQ3MTU3NjJjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDA4VDE0MTQxN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNkZjUwODkwMzdlYjlkYjI1ZDE4ODZkYjZhMjI4ZDIwNDRiZjA5ZjBiOWM1M2QyZTVlZTJiODViMTQ4MTA2MTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.GeZQQEWxqApeGbw_lfW5Jq080M6jMnmQni9KlkGv3vA)

ESPHOME uses esp-idf 4.4.6

Tested a few V4 firmwares and at one point I saw it connect but was never able to reproduce it.  At least we have an explanation for the problem.  It would be nice if it could be included in the install doc that this issue might arise with certain WiFi routers.

In my case, I'll simply disable 5Ghz on the IOT SSID

jf-navica, 2024-10-08:
@softhack007 Can you confirm that this is the proper CLI procedure?  I think my issue might lie somewhere else...

https://wled-install.github.io/bin_dir/1_Standard_version/0.14.4_V4_(ESP-IDF_4.4.3_based,%20EXPERIMENTAL,%20should%20solve%20reboot%20issue)/WLED_0.14.4_ESP32_audioreactive_V4.bin
esptool.py erase_flash
esptool.py write_flash 0x0 ./esp32_bootloader_v4.bin
esptool.py write_flash 0x10000 ./WLED_0.14.4_ESP32_audioreactive_V4.bin

Following this, it won't connect to WiFi.  But works when installing through Chromium.

softhack007, 2024-10-08:
> esptool.py erase_flash
> esptool.py write_flash 0x0 ./esp32_bootloader_v4.bin
> esptool.py write_flash 0x10000 ./WLED_0.14.4_ESP32_audioreactive_V4.bin
> NOT OK

Sorry not OK. You can't install a "V4" firmware the same way as a normal "V3" one.
A number of files are needed, which are produced when building and installing from VSCode:
* correct bootloader that matches the firmware
* app0.bin
* partitions.bin
* wled firmware.bin

I can't tell you atm where to flash all these parts. In fact I only work from the VSCode+platformIO environment, which has a nice button for "build & upload" that takes care of flashing the right parts to the right addresses.

jf-navica, 2024-10-08:
But it should work with V3 firmwares, right?

softhack007, 2024-10-09:
> https://github.com/Aircoookie/WLED/releases/download/v0.13.1/esp32_bootloader_v4.bin (as specified on the install page)

exactly what I wanted to say. The bootloader from the old WLED release pages is outdated. Don't use it.

softhack007, 2024-10-09:
As you're basically asking how to flash your board without using VSCode (the development environment for building WLED),  maybe @wled-install or @lost-hope or @dosipod might be able to help you.

<br />

And I'd suggest to move this discussion to [image]. This issue tracker is meant for reals bugs in the WLED source code...

jf-navica, 2024-10-09:
Thanks for the update.  I'll close this issue but one last question:  how to get the bootloader issue fixed on the web site?

softhack007, 2024-10-09:
allow me to repeat myself: This is not my area any more.

The webinstaller has its own repo here https://github.com/Aircoookie/WLED-WebInstaller

closed 2024-10-09 by jf-navica