wled/WLED #4191: Wrong info in json/info from audio_reactive.h addToJsonInfo

dosipod, 2024-10-15:
### What happened?

[image]
I am suspecting there is an extra char somewhere in addToJsonInfo  but yet to hunt that down 
[image]

### To Reproduce Bug

Use json/info or ws with  audio_reactive usermod 

### Expected Behavior

Correct info under  json/info or ws 

### Install Method

Self-Compiled

### What version of WLED?

0.15.0-b6 

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Take it easy :)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-10-15:
This code was last changed 2 years ago.
Can you verify if old versions also exhibit this?

dosipod, 2024-10-15:
It is indeed also in  older builds  ,went back and I can see that even in 0.14.4  
[14.4.txt](https://github.com/user-attachments/files/17377846/14.4.txt)

dosipod, 2024-10-15:
@softhack007  Hey , I thought I would look at MM to see if that is any different but it is also the same , any pointers else i will use brute force 
[MM_json_info.txt](https://github.com/user-attachments/files/17378154/MM_json_info.txt) 
@blazoncek  , Even it that is old there is still incorrect info  we just missed to report it

dosipod, 2024-10-15:
The html tags in json is expected ? if that is the case we can close this

softhack007, 2024-10-15:
Hi, yes the HTML is expected. Its needed to create the "on/off" button on the info page. I think it was never touched again after @blazoncek added it initially.

I'm not sure if JSON validation sites exist on the internet.... so the problem is invalid JSON? Or just content that was not expected by some 3rd party tool?

blazoncek, 2024-10-15:
It is a text field in a proper JSON object. Content of text can be anything, including HTML.

closed 2024-10-15 by blazoncek