wled/WLED #4193: Twinklefox looks different (less bright) since 0.15.0-b4 & "cool" checkbox is not synced

NeariX67, 2024-10-15:
### What happened?

The Twinklefox effect is less bright than 0.14.4 (or < 0.15.0-b3). Somehow less LEDs light up using the same effect parameters.
While in 0.14.4 the "cool" checkbox did not change anything, in b4 it does effect how many LEDs are lighting up (looks more like before) but it doesn't get synced to other instances in my network.

### To Reproduce Bug

Try this preset:
```json
{"on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":234,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"col":[[255,160,0,0],[0,0,0,0],[0,0,0,0]],"fx":80,"sx":128,"ix":255,"pal":50,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}
```

On 0.14.4 no LEDs turn off because of the twinkle rate and effect speed. Really good brightness
On 0.15.0-b4 **many** LEDs turn off. Low brightness

### Expected Behavior

No LEDs turn off using full twinkle rate on Twinklefox effect.
Also, "cool" checkbox should sync across all instances.

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0-b4

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-12-03:
> If I can find the time, I might try to bisect the changes to find out when it went off.

The regression is most likely due to this commit, where lots of `uint8_t` and `uint16_t` were replaced by `unsigned`.

https://github.com/Aircoookie/WLED/commit/68a7282b2765f4c43c838e7554a93a95f4385406

As a consequence, some calculations are not constrained to 8 or 16 bit any more, and might produce too-large results.

softhack007, 2024-12-03:
Similar to #4340

netmindz, 2024-12-03:
> > If I can find the time, I might try to `bisect` the changes to find out when it went off.
> 
> The regression is most likely due to this commit, where lots of `uint8_t` and `uint16_t` were replaced by `unsigned`.
> 
> https://github.com/Aircoookie/WLED/commit/68a7282b2765f4c43c838e7554a93a95f4385406
> 
> As a consequence, some calculations are not constrained to 8 or 16 bit any more, and might produce too-large results. The workaround would be to find these calculations that run into 32bit now, and add `& 0xFF` or `& 0xFFFF` respectively.

I'm tempted to say we just revert that commit - thoughts @softhack007 ?

DedeHai, 2024-12-04:
just checked: this was reported on discord back in October, when I asked for details there was no response. These changes have been in the beta for quite some time but its hard to tackle issues if not properly reported.
I am much in favour of the changes, I think its better to work out the kinks than revert the whole commit (it saves a lot of flash and may also improve speed on some FX)

netmindz, 2024-12-04:
Is this you volunteering to go through **every** effect doing the side by side regression check @DedeHai ? 😉

DedeHai, 2024-12-04:
A lot of work was spent on these changes, both in implementing, testing and fixing. I would hate to see all that work go to waste because of a handful of FX still need tweaking.

DedeHai, 2024-12-04:
fixed it. Can someone please confirm?

softhack007, 2024-12-04:
> > The regression is most likely due to this commit, where lots of `uint8_t` and `uint16_t` were replaced by `unsigned`.
> > [68a7282](https://github.com/Aircoookie/WLED/commit/68a7282b2765f4c43c838e7554a93a95f4385406)
> > As a consequence, some calculations are not constrained to 8 or 16 bit any more, and might produce too-large results. The workaround would be to find these calculations that run into 32bit now, and add `& 0xFF` or `& 0xFFFF` respectively.
> 
> I'm tempted to say we just revert that commit - thoughts @softhack007 ?

I'm happy that @DedeHai found a simple fix for the two "twinkle" effects.

I was actually advocating/pushing the "don't use 8bit and 16bit locals" change in summer. However, looking at a release to users, I agree that the "everything to unsigned" commit is causing troubles - we know about 3 regressions, and my gut feeling says, "where there is 3, there is more". 

I think we cannot test or review each effect that was touched in the remaining time until releasing 0.15.0. The added value for users is minimal (1KB saved) - 1D effects are not as time critical as 2D, due to the lower LED counts. As a compromise, maybe we revert the commit only in the 0_15_0 release brach, but keep it in the 0_15 development branch.

NeariX67, 2024-12-08:
Just got the time to test this, seems good 👍

closed 2024-12-08 by NeariX67