wled/WLED #4196: Apple Remote Codes not working with IR JSON

jay-b, 2024-10-16:
### What happened?

I'm trying to get an Apple remote (older white variant) connected to a ESP32 with a connected KY-022. Have got the codes via an Arduino sketch and created a JSON file similar to the Apple Remote v2 file from the documentation

[image]

[image]

```
{  
  "0xF40587EE": { "label": "play","cmnt": "Play/Pause", "cmd": {"ps":"1~9~r"} },
  "0xF40387EE": { "label": "menu", "cmnt": "On/Off", "cmd": {"on":"t"} },
  "0xF40C87EE": { "label": "down", "cmnt": "Brightness -", "cmd": {"bri":"~-8"} },
  "0xF40A87EE": { "label": "up", "cmnt": "Brightness +", "cmd": {"bri":"~+8"} },
  "0xF40987EE": { "label": "left", "cmnt": "Effect -", "cmd": {"seg":{"fx":"~-"}} },
  "0xF40687EE": { "label": "right", "cmnt": "Effect +", "cmd": {"seg":{"fx":"~+"}} }
}
```

I've tested using these pins: 15, 13, 4. Also, to check there isn't an issue with my file system storing the JSON file I've tested using the 24 button JSON code from the documentation which works fine when using one of those remotes. I also tried changing one of the HEX codes in that file with one from my Apple remote and that didn't work when using the Apple remote.

### To Reproduce Bug

If you have access to one of these older Apple remotes, retrieve the HEX codes, upload the JSON to a WLED installation.

### Expected Behavior

The remote should execute the JSON.

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0-b5

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2024-10-17:
Does it work if you use the codes you get from WLED over Serial output?

blazoncek, 2024-10-17:
As @DedeHai mentioned, use WLED to display IR codes, not Arduino sketch.

jay-b, 2024-10-17:
> Does it work if you use the codes you get from WLED over Serial output?

Thanks, how do I do that?

DedeHai, 2024-10-17:
connect it to USB, monitor serial, press IR keys.

jay-b, 2024-10-17:
I have the ESP32 connected via USB but see no options in the WLED settings for monitoring the serial? Sorry, very new to this project and inexperienced when it comes to Arduino stuff.

jay-b, 2024-10-17:
Managed to get the correct codes from the serial monitor in Arduino, so it's all working!

closed 2024-10-17 by jay-b

DedeHai, 2024-10-17:
for future reference: the serial output of IR codes is always active, just open the serial port with 115200 baud rate and press IR keys. If your hardware is correct, the received codes will be written to serial.