wled/WLED #4204: Alexa Presets not workong

nmeliasp, 2024-10-18:
### What happened?

When adding WLED to Alexa, presets are making it into Alexa

### To Reproduce Bug

When configuring Alexa voice assistant in sync interfaces and configure 1 or more presets, nothing makes it into Alexa. 

### Expected Behavior

All presets appear as lights in Alexa and can be toggled to turn on and off

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0-b6

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2024-10-18:
After testing with you in discord I could see this issue with the presets as far back as 0.14.2-b2 build . Since I have saved a pic of older setups testing  on 0.14.2-b2 which showed the presets working correctly I am thinking that could be related to Alexa updates.

The below is old picture but when i tested now on 0.14.2-b2 i faced the issue with presets 
[image]

Please test on 0.14.2-b2 to see if that is also broken on ESP8266 ( this should **not** work in theory  so just to confirm )

dosipod, 2024-10-18:
I could not track the Alexa updates and only found a list deprecated features  last of wish was on July 2024 for  the List API  ( this impacted others like HA ) 
https://developer.amazon.com/en-US/docs/alexa/ask-overviews/deprecated-features.html
Not really sure any is used by wled Alexa code or any of the dependencies . 

Might be contacting  Alexa Developer Support is an option but  I am not the right person to do that  :)

dosipod, 2024-10-18:
Found undocumented build flag that would do Alexa debug  (   -D ESPALEXA_DEBUG )  and I could see some errors when trying to add the light to Alexa not sure that was there as I only  used it now  but I think at this point it might be related to the same list of deprecated apis or other updates that might need a wled code update to match or other workaround 

[18:57:58]Not-Found HTTP call: /api/2BLEDHardQrI3WHYTHoMcXHgEspsM8ZZRpSKtBGr/lights/1275707904
[18:57:58]application/json
[18:57:58]FinalBody: 
[18:57:58]URL: /api/2BLEDHardQrI3WHYTHoMcXHgEspsM8ZZRpSKtBGr/lights/1275707904
[18:57:58]AlexaApiCall
[18:57:58]ok
[18:57:58]l1275707904
[18:57:58]1275707904
[18:57:58]Not-Found HTTP call: /api/2BLEDHardQrI3WHYTHoMcXHgEspsM8ZZRpSKtBGr/lights/1275707904
[18:57:58]application/json
[18:57:58]FinalBody: 
[18:57:58]URL: /api/2BLEDHardQrI3WHYTHoMcXHgEspsM8ZZRpSKtBGr/lights/1275707904
[18:57:58]AlexaApiCall
[18:57:58]ok
[18:57:58]l1275707904
[18:57:58]1275707904
[18:57:58]Not-Found HTTP call: /api/2BLEDHardQrI3WHYTHoMcXHgEspsM8ZZRpSKtBGr/lights/1275707904
[18:57:58]application/json
[18:57:58]FinalBody:

nmeliasp, 2024-10-18:
i have confirmed the same behavior on 0.14.2-b2 with my ESP8266. Where do we go from here?

dosipod, 2024-10-18:
WLED Alexa related coding was only done by AC who maybe busy with other things at the moment , 
Since your issue is still open then you would have wait as this might be a minor issue . 
I do not use Alexa with wled personally so I am just helping log the details with you but have no real interest in the outcome other then reporting bugs details in a manner that may help others to fix it  , cheers

dosipod, 2024-10-19:
[image]
This was reported a long time ago , i will try to fix that

Svennte, 2024-10-22:
Had the same problem.
I just created a pull request that should solve this issue.

nmeliasp, 2024-10-22:
Awesome! Looking forward to the fix.. As a side note I was able to get around this by exposing scenes I created in home assistant to Alexa but still looking forward to this fix

dosipod, 2024-10-22:
@nmeliasp Yes it is working and I can see the expected behavior with presets 

[image]

Change based on  0.15.0-b6  and bins are here 
https://github.com/dosipod/WLED_alexa_Oct_2024_ps_fix/actions/runs/11456536731
Please try it out on esp8266 

@Svennte Thank you  , can you re-base the PR on 0.15.0-b6 as I did in above repo

Svennte, 2024-10-22:
@dosipod 
You mean rebase to 0_15 and add the Version tag, right?
Sorry, relatively new to GitHub features. 
Will test on esp8266 later. Esp32 worked fine with Echo 4, Echo dot 5 and the new echo spot.

Norm71, 2024-11-13:
I installed the latest Binary (0.15.0-b7) but Alexa still not recognizing the presets.

Svennte, 2024-11-13:
You sure you activated it in the Sync Settings? 
Working flawlessly for me, even in Alexa routines.

Norm71, 2024-11-13:
[image]
[image]

Worked until Alexa was updated 2 weeks ago. Updated to the new version but Presets do not appear on Alexa

Norm71, 2024-11-13:
[image]

[image]

Svennte, 2024-11-14:
Working just fine for me. Did you try delete all Devices from the Alexa app and then rediscover?

[image]
[image]
[image]

Norm71, 2024-11-14:
Yes i did delete the devices and only the one is coming back all the time

Svennte, 2024-11-14:
And why do you setup 5 presets when you only have four? Maybe this is the confusing thing for Alexa

Norm71, 2024-11-14:
It was just a test and its back to 4

Norm71, 2024-11-14:
I currently try the following:
Disable sync in WLED
Reboot WLED
Delete devices in Alexa (currenty did 15 times but it still appears)

Norm71, 2024-11-14:
Now for the first time the device is called "Arbeitsplatte" as defined.
Will enable synch now again

Norm71, 2024-11-14:
Nothing changed. :( 
Devices appear also not in Device Settings

Svennte, 2024-11-14:
Do you have other WLED devices, that may inferences with this one? Like Wled Küche? 
Maybe the code won't work with multiple devices the way its structured right now and some things have to be changed for it

Norm71, 2024-11-14:
WLEDKueche is the same device but from Home Assistant. This was the same all the time. 
Do have one other WLED which does not have Segments and the Alexa Synch is off for that device.

Norm71, 2024-11-14:
If i want to trace the call with Wireshark to get the information how the call to Alexa looks like what do i need to lokk for?

Svennte, 2024-11-19:
@ChristianBracket you have version v14.04
Is it already fixed there? Sounds exactly like the problem before the update.

ChristianBracket, 2024-11-19:
I don’t know if this is just the problem I’m having. Here’s a video to show it better.
https://github.com/user-attachments/assets/93fb18ad-0d22-4cfd-9611-c1b6ab646f89

Svennte, 2024-11-19:
Please do the update first. You are using an old version which is not working anymore for amazon alexa.
Get the latest version of v15 and you should be fine.