wled/WLED #4205: JSON API returns error for "/json/live"

Cerothen, 2024-10-18:
### What happened?

When attempting to use the ```http://host/liveview``` page nothing was loading, upon closer inspection the api located at ```http://host/json/live``` was returning the following:

```
{"error": 4}
```

The "Peek" feature in the regular UI works as expected however it is communicating using websocket messages and not with the same JSON api resource

### To Reproduce Bug

Navigate to ```http://host/liveview``` when using version 0.15.0-b6 "Kosen"

[image]

### Expected Behavior

The assumption is that the ```http://host/json/live``` api resource would return a non-error

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0-b6 (build 2410140)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-10-18:
JSON live has been disabled by default in b6.

Cerothen, 2024-10-18:
Is the intent to have the live view page be removed or switch to use the websocket?

blazoncek, 2024-10-18:
If you want to use it, custom compile. Otherwise switch to websockects as it is more efficient.

netmindz, 2024-10-19:
Could we have a check during the build that catches these incompatible set of build options?

blazoncek, 2024-10-20:
If the JSON live is needed (please get users to upvote this issue) then it can be enabled in release builds.
It was omitted in an attempt to reduce binary size to fit other more important features in the future.
Actual commit: 0bbd6b7c4b677249b584f90abbacde6f44f0ea85

Cerothen, 2024-10-20:
I'm not directly aware of what's needed to address the websocket however the ```liveview``` page depends on the JSON/live API. I'm just wondering if the liveview page will be rewritten to use websockets or if it is going to be removed

blazoncek, 2024-10-20:
Web UI (including /liveview) uses websockets.
Nothing special needed.

Cerothen, 2024-10-26:
the live view page currently does try to access the json/live api.
[image]

Perhaps there is a bug?

blazoncek, 2024-10-26:
No. You have to use websockets in the parent window. UI does that.

death2all110, 2024-12-26:
Live view is still broken in the release build of 0.15.
Web UI Peek feature works fine.
Opening a new tab/window and visiting /liveview does not work. 
Not sure how I get /liveview to use websocket instead of /json/live

blazoncek, 2024-12-26:
Please read statements above. It is not broken, it was removed.
If you want to have it for your ESP, you need to compile custom binary.

amnesia106, 2025-02-25:
How can I adjust the display in the browser live view so that the 16x16 matrix LED panel is shown correctly? In the Peek preview, it is displayed correctly.

**PEEK**
[image]
**LIVEVIEW**
[image]

netmindz, 2025-02-26:
If the build only has websocket support then we shouldn't really need the user to add the ?ws modifier

Forgot my ignorance but what does the liveview give that peek doesn't? I.e why use liveview not peek?

blazoncek, 2025-02-26:
liveview is a frame embedded in UI. it can be called/invoked independently, though.
just a disclaimer: it was not developed to be invoked independent of UI.