wled/WLED #4207: Button 0 and 1 has bigger delay than 2, 3

Andro-Marian, 2024-10-18:
### What happened?

When having a button action on the BTN 0 and 1 the action is not instant than 2,3.

### To Reproduce Bug

Set a push button for the 1.
Configure a relay for the button 1.
Press one time the button.

### Expected Behavior

The reaction of the relay should be instant like the button 2 and 3.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.4, 0.15.0-b6

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2024-10-18:
WDYM "Configure a relay for the button 1" ?

Andro-Marian, 2024-10-18:
MultyRelay usermod

Andro-Marian, 2024-10-18:
[image]
[image]

Andro-Marian, 2024-10-18:
[image]
[image]
[image]
```
{"0":{},"1":{"MultiRelay":{"relay":1,"on":"t"},"n":"Relay 1"},"2":{"MultiRelay":{"relay":2,"on":"t"},"n":"Relay"}}
```

dosipod, 2024-11-01:
Possibly related , there is a report here https://wled.discourse.group/t/multi-relay-usermod-json-api/12991/9
At the end of that report we have tried the "Delay S" and it does not seem to make any difference as with or without it the relay is triggered  immediately .

Andro-Marian, 2024-11-01:
Also for my setting the delay of the relay does nothing. That option seems to be ignored.

I don't really know what the right delay is.
I think for 0, 1 is too slow, and for rest may be too fast.

Around 200-300 ms maybe would be good.

closed 2025-03-10 by github-actions[bot]