wled/WLED #4211: WS281x leds flickering every firmware after WLED 0.15.0-b3

Justdigit, 2024-10-19:
### What happened?

Leds are flickering in all kinds of colors and cant acces the webpage.
Its an Esp32 with six channels

### To Reproduce Bug

WLED firmware WLED 0.15.0-b3 works well. All versions after that not.

### Expected Behavior

Responsive webpage and normal color changes

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0-b3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-10-19:
Can you share a screenshot of your LEDs preferences? 

In case you have less then 300 LEDs per pin, AND more than 4 pins used, WLED uses a different LEDs driver (called "parallelI2S1"). It might be that your problems is related to this new driver.

Justdigit, 2024-10-19:
> Can you share a screenshot of your LEDs preferences?
> 
> In case you have less then 300 LEDs per pin, AND more than 4 pins used, WLED uses a different LEDs driver (called "parallelI2S1"). It might be that your problems is related to this new driver.

Yes thats true there are "only" 218 leds with four channels. This is because it's a PC case with RGB fans in it.
Is there a solution for it??

[image]

jonathh21, 2024-10-19:
It might not be related, but I have a matrix of 1024 LEDs.

At low brightness it goes crazy, random pixels and brightness galore.

Higher brightness less prevalent and ultimately bright enough it stops.

(I have separately observed that transitions causes instability so I have disabled it),

Justdigit, 2024-10-20:
> It might not be related, but I have a matrix of 1024 LEDs.
> 
> At low brightness it goes crazy, random pixels and brightness galore.
> 
> Higher brightness less prevalent and ultimately bright enough it stops.
> 
> (I have separately observed that transitions causes instability so I have disabled it),

Idk if it's related.... 
When I upload a firmware higher than B3 i cant even acces the webpage tp fi in WiFi credentials.

lost-hope, 2024-10-27:
Just to rule out that the flickering is a hardware issue: Do you two have a levelshifter installed?

Justdigit, 2024-10-30:
> Just to rule out that the flickering is a hardware issue: Do you two have a levelshifter installed?

No I have not.... Because prior versions just works right. They are small strings of leds.

Justdigit, 2024-10-30:
> Can you share a screenshot of your LEDs preferences?
> 
> In case you have less then 300 LEDs per pin, AND more than 4 pins used, WLED uses a different LEDs driver (called "parallelI2S1"). It might be that your problems is related to this new driver.

I have posted my config on you request.... Can you do something with it?

DedeHai, 2024-10-31:
if you reduce the number of outputs to 4, does it work correctly?

Justdigit, 2024-11-17:
> if you reduce the number of outputs to 4, does it work correctly?

No that doesnt matter. 2 CHannels or six. It keeps flickering

DedeHai, 2024-11-18:
then it may be related to your hardware. please use recommended wiring, i.e. add a [levelshifter](https://kno.wled.ge/basics/compatible-hardware/#levelshifters) to rule that one out.

Justdigit, 2024-11-18:
> then it may be related to your hardware. please use recommended wiring, i.e. add a [levelshifter](https://kno.wled.ge/basics/compatible-hardware/#levelshifters) to rule that one out.

No it's definitely not the hardware and doesn't need level shifters. 
Version B3 works like a charm. All version later not. 
The longest cable is from esp to case fan is 20cm.

Only thing I didn't try is a different esp.

DedeHai, 2024-11-18:
> No it's definitely not the hardware (case Fans and led strip) and doesn't need level shifters.

in 99% of cases, flickering is a hardware problem. I cannot reproduce you issue so please update your hardware.

closed 2025-03-25 by github-actions[bot]

mozzhead164, 2025-06-06:
I can also confirm the flickering when I updated to either '0.15.1.beta2_ESP32_Ethernet' or '0.16.0-alpha_ESP32_Ethernet' from '0.15.0-b7_ESP32_Ethernet' today. 

b7 was working perfectly until I updated. 

I am using level shifters on 2 outputs with approx 300 LED's per output (see screenshot below)
The controller is a Dig-Octa which was previously outputting smooth LED animation, have reverted back to b7 for now.

[image]

willmmiles, 2025-06-06:
Possible duplicate of #4389