wled/WLED #4212: WLED Fails to Connect When Using Web Browser

TestingWasTaken, 2024-10-19:
### What happened?

I am using the WLED app on my iPhone, and it successfully connects to the NodeMCU ESP8266. However, when I try to log in via the IP address using a web browser (on the same network), the connection fails, and I am unable to access the web interface. I have tried multiple browsers and devices, but the issue persists.

### To Reproduce Bug

Open the WLED app on an iPhone and connect successfully to the device.
Tryed logging in via the web interface using the device's IP address on a browser.
The web interface fails to load.

### Expected Behavior

I expected to be able to access the web interface through a browser using the device’s IP address, similar to how the WLED app connects seamlessly.

### Install Method

Binary from WLED.me

### What version of WLED?

Latest Version

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2024-10-20:
I am guessing the problem is with your router settings.

softhack007, 2024-10-20:
> The web interface fails to load.

I'm using the web interface very often, and never had any problems like yours. So it seems to be something specific to your home setup 🤔 like
* router settings
* wong ip address used
* which browser did you use on the PC?

Can you post a screenshot from the webUI when it "fails to load"? 

Also please be more specific about "latest version". You can find the build number and version on the "info" panel.

closed 2025-03-03 by github-actions[bot]