wled/WLED #4219: 8266 160MHz /settings/um crash/reboot loop

spblat, 2024-10-23:
### What happened?

An otherwise stable 8266 running 0.15b6 at 160Mhz crashes and reboots while a web UI is displaying /settings/um . Confirmed by ALDIY @ Discord

Demo https://youtube.com/shorts/lTwzndDkhG0

### To Reproduce Bug

Install 0.15b6 160Mhz build on 8266
Enable AR in Usermods (not sure if this is required to reproduce)
Device goes into a reboot loop

### Expected Behavior

It shouldn’t crash :-)

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0-b6 (build 2410140)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2024-10-23:
Yes that is confirmed from our side also on multiple units , esp8266 with AR on MM seems to be fine  for comparison

willmmiles, 2024-10-24:
While I've opened PR #4222 to fix the general case issue, technically this is already fixed by 26a47537.

dosipod, 2024-10-24:
@spblat We have tested on esp8266 4mb and 2mb , AR is working fine without crashes , please test the same and close this .

PR #4222 (2024-10-24): Esp8266 oappend fix
The `oappend` shim logic introduced in #4152 also has issues with `SET_F()`, as it casts away the helper type needed for `print()` to identify PROGMEM strings.  Improve the shim logic to safely handle this case in two different ways:

1. Detect PROGMEM addresses and re-apply the helper type, allowing the correct overload of `print()` to be called;
2. Ask the platform to forgive us if we make a mistake, and handle it gracefully (if very slowly).

Fixes #4219.
closed 2024-10-26 by blazoncek