wled/WLED #4224: Alexa not finding some presets and if found playing wrong one

tomware03, 2024-10-24:
### What happened?

I have 5 different ESP-32 around my room and I had to replace one due to a faulty power supply. After replacement, I set up WLED onto the new ESP-32, added all 5 presets and went to alexa to find. It found 2 of the presets but when you turn on the presets it will play a different one in WLED. As an example, the red preset is playing the green preset and vise versa. I've tried factory resetting it, fresh install and and it still cant find all of them. Its different presets everytime. Keep in mind all others are working fine and havent been touched in a almost a year.

### To Reproduce Bug

Fresh install on newest WLED update. Add 5 preesets and make alexa find them. 

### Expected Behavior

Alexa should find all presets and allow me to control them within the alexa app

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0-b6

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-10-24:
Hi @tomware03,
It looks like there is a generic problem with the alexa API -I gues that amazon changed something so WLED cannot communicate properly with the alexa servers any more (this affects _all_ WLED versions).

There is a preliminary fix in https://github.com/Aircoookie/WLED/pull/4214, but it needed more testing before we can merge that one in.

Related: #4204

dosipod, 2024-10-24:
@softhack007 Hey Maestro, We have tested the fix on esp32 and  esp8266,  ESP32S3 was tested from PR owner side  .  . The fix itself is very minor  code change and as of now there is a bug in source so that is at least better then current situation .

softhack007, 2024-10-25:
@dosipod that's good news, thanks 👍 I think we can merge the fix.

tomware03, 2024-11-05:
https://github.com/user-attachments/assets/2a10c3bd-c708-4759-89c9-cda57db661a3

So I’m still having an issue after updating WLED to the latest beta. I’ve attached a video to give you some reference as to what’s happening. Would appreciate as much help as possible.

tomware03, 2024-11-06:
I've updated all devices and currently they are working fine and alexa is able to find my presets. Thank you very much. Will let you know if I come accross anymore issues

closed 2025-03-13 by github-actions[bot]