wled/WLED #4228: Scrolling text stops scrolling after relais switch off

robertvb83, 2024-10-26:
### What happened?

I have a matrix panel with eight 16x16 panels arranged 4x2

Relais is set to i/o pin 5

I use a poti to control brightness as well as a switch set to an on/off preset.

I setup
a background segment 0-64 , 0-32 (total panel)
A top segment 0-64 , 0-16 with long scrolling text overlay
A bottom segment 0-64 , 16-32 with #DDMM

Everything works fine if this preset is started (also if started at boot)

However if poti is turned to off so that relais cuts power to leds or if the switch is set to off, the text will freeze and no longer scrolls when lights turn back on

### To Reproduce Bug

See above

### Expected Behavior

Text should be scrolling through power cycles of the relais 

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.15-b3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-10-26:
Please use b6 and report back.

blazoncek, 2024-10-26:
It will resume scrolling after a while.

robertvb83, 2024-10-26:
I self-compiled version as of today. The issue still persists

blazoncek, 2024-10-26:
Can you confirm that changing `led.cpp` function `toggleOnOff()`to this:

```c++
void toggleOnOff()
{
  if (bri == 0)
  {
    bri = briLast;
    strip.restartRuntime();
  } else
  {
    briLast = bri;
    bri = 0;
  }
  stateChanged = true;
}
```
fixes the issue?

robertvb83, 2024-10-26:
It fixes the issue for on/off via switch!
On/off via poti still results in text freeze... Does that make sense?

closed 2024-10-26 by blazoncek