wled/WLED #4230: Build flag WLED_DISABLE_ADALIGHT will break the settings/sync

dosipod, 2024-10-27:
Compiling with  WLED_DISABLE_ADALIGHT will break the settings/sync page and all the values in other areas will be empty.
This was part of the testing for c3 but it is also the case on classic esp32 
[image]

blazoncek, 2024-10-27:
Disabling Adalight will also disable Improv and all serial communication. It is not really intended to disable Adalight.

Though there is a bug when it is disabled.

dosipod, 2024-10-27:
We were looking for anything that could reduce the bin size on c3 to fit audio and maybe another usermod and below is the most common feature we could remove without too much impact 

-D WLED_DISABLE_ADALIGHT 
  -D WLED_DISABLE_HUESYNC 
  -D WLED_DISABLE_LOXONE 
  -D WLED_DISABLE_INFRARED
  -D WLED_DISABLE_ESPNOW

We do intend to do more tests for ALL other build flags even if not related to size  just part of 15 testing if you see that to be useful but if anything is related to functionality testing after using a build flag then might take a bit of time . I am on mobile so I hope this is clear

closed 2024-10-27 by blazoncek