wled/WLED #4232: Add date and time into wled-ci.yml output files for beta testing

dosipod, 2024-10-28:
This is not an issue but opened here due to "Discussions" not being available **which it should**  as some of the active devs do not visit discord and some of the questions opened as issues are closed without clear answer  other then that it is not a bug or not planned  . If discussions is enabled then that would be a community effort that does not necessarily need a dev involvement . 

To start it off I am trying to add date and time  to the builds both zip and bins  , this is needed  as  I am trying to test a lot and  I get confused as to which files to use as the file name generated from github is having the same name  , would also help us go back to specific date or time we built in  , I tried to add that with date variable to  wled-ci.yml but I could not get that to work  correctly .  The https://wled-compile.github.io/ yaml has date example but written to a file if that is a good place to start.

This is just for us to help test in a more organized way and might be specific question to  @willmmiles or @softhack007  .
Otherwise I might try to bug @wled-install  for it 

Cheers

willmmiles, 2024-10-28:
To confirm my understanding of the problem you're trying to solve:

- By default, the output binary name in the `build_outputs/release` folder lists only the "base" version, and target environment.
- This creates two issues:
  - It is not possible (or at least impractical) to distinguish "clean" builds of the release tag from development or testing builds with additional patches
  - The filename does not capture which options are enabled without creating a custom environment
- This problem is made worse when using Github codespaces or CI for development, as the build output files collect in the downloads folder with unhelpful filenames, like `WLED_0.15.0-b6_ESP32_WROVER (3).bin`.

Is this correct?  I've definitely had trouble with keeping track of which build came from where, particularly when I'm slinging a bunch of experimental branches around.

Sorry if I seem like I'm being a bit pedantic - I want to make sure I fully understand the problem before looking at solutions.

willmmiles, 2024-10-28:
There are a few problems that will come up with just straight embedding build timestamps in variables and filenames:

1. Builds are not reproducible anymore -- if you check out the same source code, you get something different.
  - It can be worse than just "some string is slightly different" - changes to the string length can affect the layout in memory, and that can impact how many instructions are needed to load things in functions... it snowballs fast
  - This can be somewhat mitigated by disabling this behaviour for release builds
2. Unbounded growth of output folders.  If you do a lot of compiles, you can end up with hundreds or thousands of old builds kicking around that need to be cleaned out periodically.
3. It'll break PlatformIO's object file caching.  If the build data is passed in via a compiler flags, eg. `'-D VERSION=my_build`, it'll be recompiling everything all the time.
4. Build timestamps actually tell you basically nothing about the build.  I don't always remember which order I built things in, I don't know about anybody else.

In the past I've built something based on parsing git metadata, eg. `git describe` -- it'll tell you the branch or tag, how many commits past it, give the actual commit hash, and indicate if there are uncommitted changes.  This solves problems 1 and 4, above, with the caveat that if you ever care to roll back or re-read some specific code, you need to commit it first.  (This is best practice anyways as debug code can be squashed away later).

I'm pretty sure we can work around problem 3 with some extra logic in the platformio hooks.  We'd probably want to isolate the version data to its own object file, and then override the build flags just for that one file.

I can live with problem 2 being an exercise for the developer for the time being.   It won't bother CI, at least.

I can look at this in more detail in a week or two.  I am somewhat backlogged with halloween stuff at the moment.

DedeHai, 2024-10-28:
If the issue is only the file names: isn't there the option to run a post build script? Could that be used to just rename the files and add a timestamp? that would leave the actual build untouched.

blazoncek, 2024-10-28:
I've also experimented with `git rev-parse --short HEAD` which will give you commit hash, which can be good as you know from which commit it was built from, but it cannot solve every aspect.

As for filenames, `WLED_RELEASE_NAME` is the one to modify if you want different binary name. But again that is static and you cannot tell which build it was from.

netmindz, 2024-10-28:
I'll try and send a longer explanation later, but in short I would say do not change anything about the actual build itself, only the naming of it artifact that are produced by the CI, I typically name these either based on the githash for non-tags and also include the branch name too

willmmiles, 2024-10-28:
> I'll try and send a longer explanation later, but in short I would say do not change anything about the actual build itself

I'd definitely like to hear more about your experiences here -- I was really hoping to embed the build details in the binary, so we can definitively map debug reports to specific builds.  I use Ghidra a lot for digging through stack traces and core dumps, and matching the exact binary used is critical.

closed 2024-11-05 by dosipod