wled/WLED #4235: WLED - Unknown Error when attempting to add new integration one Homeassistant

boyi01, 2024-10-30:
### What happened?

I upgraded from an old 0.12.x ig uess to 0.14.4
I tried to re add the in the integration in HA, no chance i get this error

### To Reproduce Bug

I have no idea, i can reproduce by try to add it again

### Expected Behavior

get a valid ha integartion

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.4

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
wled.exceptions.WLEDError: (404, {'message': '<!DOCTYPE html><html><head><meta charset="utf-8"><meta \ncontent="width=device-width" name="viewport"><meta name="theme-color" \ncontent="#222222"><title>Not found</title><style>\nbody{font-family:Verdana,Helvetica,sans-serif;text-align:center;background-color:#222;margin:0;color:#fff}img{width:400px;max-width:50%;image-rendering:pixelated;image-rendering:crisp-edges;margin:25px 0 -10px 0}button{outline:0;cursor:pointer;padding:8px;margin:10px;width:230px;text-transform:uppercase;font-family:helvetica;font-size:19px;background-color:#333;color:#fff;border:0 solid #fff;border-radius:25px}\n</style></head><body><img alt="" \nsrc="data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAACAAAAAgCAMAAABEpIrGAAAAAXNSR0IArs4c6QAAAARnQU1BAACxjwv8YQUAAAAbUExURQAAAAB81gCU/zKq/////9bW1oCAgGhoaAAAAGPLX8AAAAAJdFJOU///////////AFNPeBIAAAAJcEhZcwAADsAAAA7AAWrWiQkAAACdSURBVDhPxc9bDoUgEANQebP/FUuHMjBGY/B+3EYR7RH0qC/ZBc6HwCljgHO+xZIVSI2sYgHaG7EBWh8jWoxTrCBFdDJ+BD4lbIHxAcz8APAVLTsrZE4eQD5qzt3cAFTYokC4YCN9Gybgu4yAQtBFLQXHuHABA7JMeOEC/E0W5uy9gv4vo5QHK2i7yq2C8UABM4HmL+CSTXCTF1DrCX6+Gp9zB5dsAAAAAElFTkSuQmCC">\n<h1>404 Not Found</h1><b>Akemi does not know where you are headed...</b><br><br>\n<button onclick=\'window.location.href="../?sliders"\'>Back to controls</button>\n</body></html>'})
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-10-31:
> 404 Not Found
> Akemi does not know where you are headed...

This is the "page not found" 404 message from esp webserver. There must be an error in the HA integration, because it's trying to use a URL with errors (-> page not found)

dosipod, 2024-10-31:
I just added esp8266 running 0.14.4 without an issue . I see few issues like that in the past with HA and it needed a cleanup 
[image]

boyi01, 2024-10-31:
Yeah i guess it has todo something with ha integration, the app works...

is their a better place ?

i have recreatet the ha container... nothing changed

softhack007, 2024-10-31:
@boyi01 the place to report homeassistant bugs seems to be https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+wled%22

boyi01, 2024-10-31:
I'm just stupid... at first glance it looks like this is only for the core..

closed 2024-10-31 by boyi01