wled/WLED #4240: Preset turning off when turning off light

finity69x2, 2024-11-01:
### What happened?

If I go to the dashboard and select a preset then turn on light everything is good

but when I turn the light off the preset gets deselected.

this happens from the WLED dashboard and in home assistant

this really screwed up my home assistant automation since I use the preset selection to control automations.

### To Reproduce Bug

select a preset
turn on light
turn off light
preset gets deselected and shows as unknown in home assistant

### Expected Behavior

the last preset should stay selected when light is off

### Install Method

Self-Compiled

### What version of WLED?

0.14.4 and 0.15.0-b7

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

it worked as expected in some previous version of WLED. I was forced to update to 0.14.4 as the minimum version when updating home assistant. I'm pretty sure that it worked as expected in version 0.13.3.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2024-11-01:
what do you mean by "preset gets deselected" ?  when you turn off the leds, the preset just gets paused, it is not unloaded. this looks like a problem on HA side to me.

finity69x2, 2024-11-06:
It's not a HA thing.

it happens whether it's controlled from HA or from the WLED control panel.

here's gif to show what's happening:

[image]

as you can see when I start out that the preset in both HA and WLED shows as 'yellow'.

I then turn off the light from WLED and the 'yellow' preset turns off in WLED and the preset in HA goes to 'unknown'.

but you are correct that WLED does remember the preset somewhere because if the light gets turned back on from either HA or WLED the color is still yellow from the last preset selected.

it just doesn't indicate anywhere that I can see that it's the case.

DedeHai, 2024-11-06:
I see. what parameter is that you are reading from wled that gets disabled?

finity69x2, 2024-11-07:
I'm not sure I understand the question...

the preset gets disabled in both HA and WLED.

I don't know how HA determines what the preset is but that is a built-in function that HA reads from WLED. Once the preset gets turned off in WLED (or whatever is happening) then HA just reflects that.

If that's not what you are asking then could you please clarify?

DedeHai, 2024-11-07:
the question is how does it determine that it is not selected, i.e. what "value" or parameter is requested from WLED.

netmindz, 2024-11-07:
Looks more like a feature not a bug. If you are turned off then by very definition surely there is no actively running preset? Did any version behave differently? If so which version of WLED worked as you expect?

With regards to automations, please can you explain this a little further

finity69x2, 2024-11-07:
as you can see at the beginning of the gif sequence above the 'yellow' preset is selected. it is the one that's highlighted.

and in HA it shows as 'yellow'.

once the light is turned on or off the preset in WLED gets unhighlighted (deselected) and the preset in HA goes to unknown. 

it doesn't get re-selected if the light is turned back on either.

here is another gif example but using the white preset instead of yellow:

[image]

as you can see HA is simply reporting what WLED is doing.

And I can then re-select the preset in HA and it shows as selected in WLED.

I'm pretty positive that this didn't happen in my previously installed version of WLED (0.13.3). it only started when I updated to 0.14. as a requirement for HA to support WLED.

closed 2025-03-15 by github-actions[bot]

omayhemo, 2025-10-25:
This is still an issue.  WLED need to remember presets between on and off.

blazoncek, 2025-10-25:
> This is still an issue. WLED need to remember presets between on and off.

No. Preset (by default) is a saved state. If you change anything you are no longer in that "saved" state, hence deselection of preset.

willmmiles, 2025-10-27:
I think that HomeAssistant may have a different idea of what a "preset" means than WLED.

To WLED, a "preset" is a "set of instructions for changing the system state".  A preset can change pretty much anything in WLED, and it isn't required to supply a complete state -- the easy case is a preset that effects only certain segments, but presets could do things like "just change the color without changing the selected FX".  Or you could write a preset that does "start switching between 8 other presets in random order".   A preset is basically a canned set of commands to WLED; they could be anything (or even nothing).

And that's really the key: a WLED preset *does not define a specific state* -- it only defines *a set of instructions to make changes*.  You can build a preset that completely specifies the output state, but this is not required.  So to WLED, there's no real concept of "being in a preset" -- a preset doesn't always define enough to be a real state.  All WLED can report is that "the last preset someone executed was 9, and no changes have been made since".  So when HomeAssistant says "WLED is in preset 9", that's stretching the truth a bit -- the preset number doesn't guarantee a specific LED state.

As another example of how things can break down, consider a WLED instance driving two segments.   Presets 1-9 have ben set up to configure only segment A.  Presets 10-19 have been set up to only configure segment B.  This is a perfectly valid (even common!) WLED setup, but HomeAssistant can't handle this.  Consider: the user first activates preset 8 to set segment, A then preset 16 to set segment B.  HA has no way to show "both 8 and 16".

As a matter of being truthful, whenever *any* change is made, WLED is no longer in "whatever state we landed in after executing preset 9", so it clears the reported "preset" field.  And since WLED cannot assume that all presets fully define an end state, it also doesn't try to compare "is the current state the same as some preset" after any manual change, as that could potentially yield many answers (even apart from being expensive to compute) .

So ultimately I think this is maybe a bit of a mismatch between what HomeAssistant (and many users) think of as a "preset" -- a specific, fully defined state of LED outputs (but not including an overall "power" state!!) -- and what WLED thinks of as a "preset" (a script of arbitrary instructions).

----

All of that said:  From a code perspective, I think there's probably an easy change that would satisfy most users -- WLED could always explicitly report two fields: "last preset executed" and "state has been changed since preset".  In the case where people are fully defining the state in presets, this will let HA do what most people seem to want - it can map the "last preset" field to its idea of "current preset", and ignore the "state changed" bit.   Anyone setting up partial-state presets with HA integration will have to accept that the "last preset" field is only what it says on the tin.

blazoncek, 2025-10-28:
> downgrading to 0.13.3 helps others

AFAIK recent version of HA no longer supports 0.13.

softhack007, 2025-11-14:
> If you are turned off then by very definition surely there is no actively running preset?

This is actually how it was designed ("meant to be"). If you change segment options or global brightness or on/off status, the active segments are different from the last preset selected, hence the preset is not marked any more in the WLED UI, and HomeAssistent concludes that the active preset is "unknown". 

Technicially everything is correct: after "lights off", what you see is a "temporary preset", not the original preset from the list. So the preset is not marked any more. Internally, WLED is running a temporary preset with ID=255.

VMmatty, 2025-11-14:
> This is actually how it was designed ("meant to be"). If you change segment options or global brightness or on/off status, the active segments are different from the last preset selected, hence the [...]

It can still be technically correct and also exhibit behavior that is either not ideal or just plain broken.  This worked perfectly up to 0.13.3 and then integration between HA and WLED has been "broken" since then.  If you don't use multiple segments then you probably have never even noticed this.  But if you use WLED integrated with HA and have multiple segments and multiple presets then you end up in a situation like me where the lights become unknown in HA and the only way to fix it is to go into WLED and turn the preset back on.

I understand from this thread that according to WLED this is working correctly and that this is more of an HA problem.  I accept that.  Still disappointed in where we are.

christerdk, 2025-11-15:
Since this issue is stale and marked wontfix, I hope it's ok I dive one step further into this from a Home Assistant perspective and share my two cents on how I solved it in my context.

This was an issue for me because I used the current preset on the WLED integration as the source of truth. So while possible in the WLED UI without side effects, as soon as I changed the master brightness from HA, the preset state went unavailable (or was it unknown?). The above hints about why.

What worked for me was to take over the responsibility of which preset was expected to be active at a given time. I use a an Input Select for this. This means that even if I make slight changes (same preset, but different brightness), the preset setting might change in the WLED controller but the helper in HA remains the same. 

Now I can combine time based presets (morning, evening etc.) with event based presets (doorbell, media streaming) _and_ adjust if necessary, but never loose track on the intended active preset, which I can re-apply if desired.