wled/WLED #4242: Segment defaults to 22 after a power failure

Ace0510, 2024-11-01:
### What happened?

I have a segment with 260 leds, but every time we have power failure it defaults to 22. 
v0.14.4 “Hoshi”
Home Assistant 
Core
2024.10.3
Supervisor
2024.10.3
Operating System
13.2
Frontend
20241002.3

### To Reproduce Bug

every time we have power failure

### Expected Behavior

Keep the data intact 

### Install Method

Binary from WLED.me

### What version of WLED?

v0.14.4 “Hoshi”

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

No

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2024-11-02:
Do I dare ask @Ace0510  , Have you actually saved a preset  after you setup your segments in WLED UI ? If you did then remove HA from the equation as you could do anything with HA and that includes the good and the bad  .
Otherwise provide more info so we could try to reproduce

Ace0510, 2024-11-02:
Thank you for the quick response. It’s true that the first time I did not
save it, but I could see it once I opened it again. To make sure  I am
doing it correctly.  I change the stop led to 260 then I click on the check
mark on the second line.  I do it on the WLED App. I only use HA to turn
them on/off and change the effects/colors.
I have two other ones with ESP32 and no issues with them. This one is an
old ESP286 mounted onto a Quin-dig-uno. Any other suggestions?
Thank you again,
Alberto

closed 2025-03-04 by DedeHai