wled/WLED #4247: Palette effect glitch

DoreK333, 2024-11-03:
### What happened?

After updating WLED from 0.14.4 to 0.15.0-b7, the Palette effect from saved presets stopped animating. When I re-select the Palette effect and some colors, the animation is very glitchy and does not work properly.

### To Reproduce Bug

WS2815

1. Update to 0.15.0-b7
2. Select Palette Effect

https://github.com/user-attachments/assets/7927770c-80fe-4f91-9251-bc8f556db6fa

### Expected Behavior

Smooth animation of the color palette as before the update

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0-b7 (build 2410270)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2024-11-03:
did you disable the rotation checkmark?

DoreK333, 2024-11-03:
now yes, that was the reason, I didn't notice it before :)

closed 2024-11-03 by DoreK333