wled/WLED #4248: Pixart fails with syntax error with grid size greater than 58x32

robertvb83, 2024-11-03:
### What happened?

I have a matrix 64x32 pixel (4x2 16x16)

I can generate pixel art with the pixart generatir, however simulation or saving of the preset fails if dimensions are set any higher than 58x32.
E.g. 58x32 works, 32x32 works, 55x32 works. 59x32 fails 60x32 fails 64x32 fails

There is a error message if it fails: SyntaxError: unexpected end of JSON input

See below a failed JSON output

[image]

### To Reproduce Bug

See above

### Expected Behavior

Pixart should work up to my max matrix resolution

### Install Method

Self-Compiled

### What version of WLED?

WLED v0.15.0-b6 2410260

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
{"on":true,"bri":53,"seg":{"id":0,"i":[0,2,"000000",2,4,"d2f4ff","d3f5ff",5,10,"d2f4ff",10,12,"d3f5ff",12,16,"d2f4ff","d3f3ff","d2f4ff","d3f5ff",19,22,"d2f4ff","d3f5ff","d3f3ff",24,29,"d2f4ff",29,35,"d3f3ff","d4f5ff",36,38,"d3f3ff","d3f5ff","d3f3ff","d4f5ff","d3f3ff","d2f4ff",43,45,"d3f5ff",45,61,"d2f4ff",61,66,"000000",66,68,"d1f3fe","d2f4ff","d1f3fe",70,72,"d3f3ff","d2f4ff",73,76,"d1f3fe","d2f4ff","d1f3fe",78,82,"d3f3ff",82,87,"d2f4ff","d3f3ff","d2f4ff","d1f3fe","d3f3ff","d1f3fe",92,96,"d3f3ff","d2f4ff",97,99,"d3f3ff","d2f4ff",100,106,"d3f3ff",106,108,"d2f4ff","d1f3fe","d3f3ff","d2f4ff","d3f3ff","d2f4ff",113,117,"d3f3ff","d2f4ff","d3f3ff",119,121,"d2f4ff","d3f3ff",122,124,"d2f4ff","d1f3fe",125,130,"000000","c3e6ff","a1cbe2","c7edff","b4dae9","b7ddee","a0c4e1","b3dbf2","d9faff","c0e8fd","a4d0ea","9fc7e7","b9dcf2","c7edff","d8fdff","bbe1f4","bae3f5","c3e9fa","d7fcff","c5ebfe","b7ddf0","cef6ff","a5cae2","c8ecfd","cef2ff","aed1f3","a7c9e7","a3cae2","c2ebf9","a5cce8","c4e8fc","b4daeb","ccf4ff","b7e0fe","add6f1","9fc7e0","c7f0ff","a9d1e1","b3dbf0","bfe8f7","d3faff","c1e4f8","9ec8de","c2e8fb","bdeafb","a0c9e8","c7eaff","c7efff","c2ebfb","afd5e8","c2e8fb","cdf4ff","9fc4dc","a9cee6","d6f6ff","b2d4ef","d5faff","a2c9e2","c4e9ff","b2dcf0",189,194,"000000","cfeffb","f4ffff","caf0ff","c9efff",198,201,"c7eefe",201,203,"c7f0ff","c6effe","c7f0ff","ccefec","cdeaf0","bcbd58","b3b65f","c5f1ff","caf0ff","c9efff","c7eefe","caf0ff","e2f3fb","c9efff","c7eefe","fdffff",218,220,"ffffff","c9efff","c7f0ff",222,224,"c7eefe","c6effe",225,228,"c6eeff","c6effe","c6ecff",230,232,"c6edfd","c5edfd",233,235,"c5edff","c6eeff","c5edff","c6ecff","c6effe",239,241,"c6ecff","c5edff","c7eefe","c9efff",244,247,"c7f0ff","c9efff","fffcfe",249,251,"ffffff","caf0ff","c9edff",253,258,"000000",258,260,"ffffff","e2faff","c4ebf9","c4ecfc","b0d6eb","a9d0eb","d4f6ff","a4cfee","b9bd51","8fac59","234c57","b2bc60","3d6248","1d4750","1c464e",274,276,"ffffff","cff1ff",277,279,"ffffff","eefeff",280,282,"ffffff","fdffff","cdf1ff","dff9ff","c3e6fd","cff4ff","b7deec","c0e3f9","b3dcec","bae2f9","aed7f2","b4ddef","c7eefe","dfffff","a3d0ec","d3f2ff","cdf0ff","b8e0f2","d1eeff","aed9f6","c5ecff","bde3f6","c1eafc","b0d8ec","cff3ff","adcfeb","bddef0","d5e8f1","fcfeff","cbf0ff",311,313,"ffffff","d4cb73","204e4e","2f5f51","3a675f",317,322,"000000","162e41",323,325,"ffffff","fffffe","dcf5f6","528b6d","90ad76","91ad62","195359","195a5d","729a66","2a5c52","21474d",335,337,"1c464e","1a434c","ffffff","f0fbff","d7f7ff","cdf1fe","cef2ff",343,346,"ffffff","cdf1fe","cdeeff","dceff4","fdffff","cff2ff","d7eefb","f2ffff","f0fcff","d1f1fd","cef2ff","cff4fc","cff4ff","cef6ff","cdf4ff","f6ffff","f3ffff","f4fdfe","ffffff","d3f1fb","eafbff","d4f2fc","d4f4ff","cdf1fe",369,373,"ffffff","effafb","f4feff","ffffff","fffff9","6e985a","265557","1f4c4d","8aa659",381,386,"000000","153243","153040","142f3f","ddebe9","d3e4e0","d4e9e4","cee8d6","578563","e9e3b5","9aa650","739b60","1c4547","1a4051",399,401,"1b424e","1a414b",402,404,"19404a","cef3fd",405,409,"ffffff","d5f3fb","d1effc","dff4ff","cdf1ff",413,416,"caf0ff","ffffff","d8f0f8","d4f2fc","cff4ff","d0f5ff","d2f4ff","ffffff","d0f5ff","f9ffff","e9f6f8","d4f1fe","d3e6e0","396665","447b76","2a5f5d","437979","ffffff","215057","214e55","fbfffe","f8ffff","fbfdff","ffffff","fffffb","437048","6a8b5b","194242","4c7555","1d4a4d",445,450,"000000","133343","163143","153040","142f41","b4c9d4","dae9d4","f8f7d3","69866a","1e545b","cad76a","437662","396a52","2b5657","2d5961","2b5661","224a53","365d67","305a62","255159","d1f1fd","cceefc","cdf1ff","cbefff","cbf0ff","caf0ff","cbf0ff","fdf6cf","9fae60","306c66","316a67","316a69","f0fbff","e0f4fb","def8fd","cff4fe","d2f4ff","d0f5ff","feffff","fffeff","cbf0fd","cef0fe","648774","295e5c","447e7a","285e5e","fcfffe","ffffff","f6fcff","e7f3ef","719456","799d4d","547c65","2b483a","c0c958","426247","90a04d","3e6c4f","1f4952","1c4e4f",509,514,"000000","113544","163143",516,518,"153040","162e3f","dde8eb","d6eae1","9cbcb1","5b8c83","84b09f","b4cb74","4a7f71","4c8761","356b62","2e6356","3c7269","3f7965","31645e","3a6661","3c6869","1b434c","d3f6ff","d3f5ff","bde0f9","d1f7fc","c4c068","d8d174","2e6960","2e6967","2d6866",544,546,"2e6768","fbffff","e9fbfc","f2ffff",549,551,"ffffff","c9f2ff",552,554,"cbf2fe","316364","39726f","55868a",557,559,"ffffff","fcfffe","4f7964","1b4953","1a474c","1b463c","1a4950","d2c95a","dbc974","6a8c53","153b4e","1d4c51",570,572,"215055","a2ab54",573,578,"000000","113844","1d4752","0d3442","153042","142f3f","142d3e","122d3d","c1cfcb","c9dcbd","82a99e","b7c978","b8cb75","457a76","447c70","488066","4a8463","498169","417e67","366f5e","3f7963","346959","1c474a","abb76f","718e67","617f7a","d4d078","97a86a","619d9b","528b8e","43787f","2e666c","377275","2c6566","2f6869","81b8ba","83b8bd","84b7bf","85bac1","8abfc8","cce9ed","ffffff","fcfffe","578e8e","fcffff","d4f2fc","193e49","204751","1b4850","1c464e","d2d596","a9ad41","1b4d4c","879a58","1d4c53","849e56","2b584f","9fa857","a0ac5f","6c7741",637,642,"000000","336655","cfb43e","2f5c5f","1e414a","153042",647,650,"142f41","102c43","e5f0e1","bfcf88","b2c588","9cc177","96c276","548a65","55896f","4d8a65","3e7469","2e6459","2b5e58","488164","498064","1a434c","18444c","e8e6be","e8dfa6","bfbd75","7fb3ba","80b6bd","7eb3ba","79afb6","72aab0","74acb0","7cb1b8","6da8a6","ffffff","669a9d","efffff",680,683,"ffffff","fcffff","07132d","0a213b","153042","183846","1b434a","d5d17a","ecdb73","437f51","a9ad41","1c4e4d","cbb946","a7a83f","2f5a42","8ca256","637939","b5ad42","497755",701,706,"000000","dbc851","dac74c","d7c64f","d5c650","ddce6a","efe4b6","f6efd3","485646","142d3e","162c3e","f5ffff","efffff","92b7b3","aad5cd","6ba291","467a6f","3a6e58","5a8778","518772","306256","4f8467","598979","27564d","3c6b62","44726e","e8fcff","ffffff","e8f7fb","f8fffd","eaf7fb","bcd7e3","d4e6f0","c0e2f2","ecf8ff","fbffff","beddf4","c4e3f6","162834","49556a",745,747,"162e3f","172f40","172f42","153241","c5b83b","c0ae3e","598549","c0ae3e","4e7951","b3b442","8a963c","5e8652","183549","647e31","162124","17082d","557d40","274a38","294242",765,770,"000000","81963f","c3b538","9c8e2c","c0ad34","bca92e","0a0a0c","748a3a","435644","6e781e","95983e","8a8d38","beb843","e5ecce","e7f5f5","cfdddb","cedecd","aecebc","45795e","5d9469","477b52","437b66","518966","4c845e","568860","57865d","578561","548174","d3ebf0","f1feff","cbe1ee","f6ffff","dff8fd","ecfbff","d3eaf4","d5ecf9","f4ffff","163141","153243","163342","163143","c1b334","526e42","c2b13a","7a8143","9ca959","457048","c7b73b","51754b","a6ad3e","2b5437","18072b","150329","160021","c7b545","1b2828","27442d","130022","11021b","110120",829,834,"000000","445438","cdb436","677b42","cfb231","cab131","c2b13e","bdb44e","cabd46","c9b236","0b122b","43653e","0b0220","1a4434","162930","1e383d","88944c","e4f0da","b3bb6f","9abe79","7bac79","73a677","6ea577","61966b","508063","598863","5a8963","5a885d","5c8a5d","5e8a5e","5d895d","5b875a","264f3f","173743","153042","163344","5e855c","688b49","8d974f","b1b053","517947","8e9c51","335d46","adac44","83984d","0f102b","150427","080d22","110321","130020","09050f","a4a943","15434d","18143a","140226","130020","10001e","13011d","5a420f","4f7c47",893,898,"000000","46584b","9e983d","4d6b41","7e8c2a","636e48","657937","103134","85974f","7b7921","516d41","97933c","170629","234336","57843e","130020","120420","101b37","192f33","202f28","545b2e","b0e2f3","aee0f1","afe1f2","b0a4c1","aed6dd","aad1df","a9d2df","a6d3e1","a6d2e5","a4cee0","9dc8d5","cecbe0","b8e7f9",931,933,"b4e5f7","b0e2f3","a6d7e9","a2d4e5","98cfda","70a3af","164450","1f5059","113f48","19404c","173c48",943,945,"173949","173645","133341","143142","112f3e","162e3f","b9a62f","497045","4d7148","9c942c","66894c","657a31","c2ad3c",957,962,"000000","24353d","292a3e","10082a","1f2d2b","182b35","2d402e","c0ad34","bba433","091 [... cut ...] f5f","1a132f","12051c","13011d","110120","0e001a","0f041f","130020","33374a","738981","5a7f6e","140020","130020","110120","130020",2016,2018,"13011d",2018,2020,"130020",2020,2022,"13011d",2022,2024,"10001e","110120","10001e","10001c","6f9043","749046","749044","749042","9d9e47","748e41","738f43","758f44","729144",2036,2038,"6f9045",2038,2040,"6f8f49","6e8e4a","6f8f4b","6e904b","6d8e4c","6d8f4a",2045,2048,"000000"]}}
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-11-03:
Individual pixel control AKA `{"seg":{"i":[...]}}` has limitations when sending large pixel counts.
Split request into multiple smaller requests.

softhack007, 2024-11-03:
Another workaround - if you can live with "blocky" images - would be to set grouping=2 in segment settings. It means that a 4 pixels square will be treated as a single pixel. So to fill your 64x32, you only need an image of 32x16.

closed 2025-03-13 by github-actions[bot]