wled/WLED #4251: Push button is being ignored and do nothing

idiamant, 2024-11-05:
### What happened?

Connected a push button to GPIO 25 and GND. Trying to work with it, and it does nothing. Tried using Button 0, Button 1, 2 and 3 also, tried using the default settings and a preset I made for toggling the power state, and nothing works.

I'm using ESP32. Tried versions 0.14 and 0.15, b4 and latest b7, nothing works. Before using GPIO 25 I tried using another GPIO for that (don't remember which one) and it also didn't work. Chose GPIO 25 now after seeing some tutorial suggesting to use that one on ESP32.

Didn't find any way to debug in the WLED itself to see if it recognizes the press or not.

### To Reproduce Bug

Connect a momentary push button to GPIO 25 and GND in the ESP32. Define Button 0 or Button one as GPIO 25 and Pushbutton.

Trying to click the push button, nothing happens. Creaeting a preset, and defining that preset in the Button's preset config, nothing happens as well.

### Expected Behavior

I expect to see the WLED react to the push button press, and toggle the light mode.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0-b7

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-11-05:
@idiamant did you take a look into the WLED Knowledge base already? It contains lot of information about how to configure WLED.
https://kno.wled.ge/features/macros/

In you case  - you did not add a pullup resistor - it might be that "Buton inverted" must be used.

idiamant, 2024-11-05:
Hey, thanks for the reply. I didn't use a pull up resistor. However I did succeed eventually to make it work by switching the GND pin to a different one in the board. No idea what's the difference between these, but now it works.

Thanks anyhow for the reference to this part in the doc.

Closing the ticket.

closed 2024-11-05 by idiamant